shaise/FreeCAD_FastenersWB #124: Unable to opem fasteners after latest update

Spiralis, 2022-02-09:
I think this is for V0.3.37, but since I am not able to start the module it is hard to know. I am running on Windows 10.

When I start FreeCAD both v19.3 (stable) and the realthunder (stable) versions behave the same,

When I go to the add-on-manager and try to load the Fasteners workbench I get this error:

[image]

With this logged in the console:
```
14:57:31  Traceback (most recent call last):
  File "<string>", line 53, in Initialize
  File "C:\Users\ronny.GLOBALINTS\AppData\Roaming\FreeCAD\Mod\fasteners\FastenerBase.py", line 114, in FSGetCommands
    return FSCommands.getCommands(group)
  File "C:\Users\ronny.GLOBALINTS\AppData\Roaming\FreeCAD\Mod\fasteners\FastenerBase.py", line 89, in getCommands
    for cmd in self.commands[group]:

14:57:31  ('screws',)
```

I originally had the module working, but I had issues with tap and die for custom diameters with 3d printer comaptibility on. I found that there was a newer version and upgraded. But, since the toolbar was so cluttered I changed the setting in fasteners to dropdown, as I thought that potentially could create a less crowded toolbar. I had not tested this option before, and nor do I know if it is related to the issue of fasteners not loading at all. 

I have tried to uninstall and reinstall the Fastener add-on without luck. I have also tried to remove the Fasteners module in the Mod folder manually, as well as at the same time removing any references to Fasteners in the link.user.cfg (in the AppData\Roaming\FreeDAC folder). which is the only place I found any config-references to the Fasteners workbench.

shaise, 2022-02-09:
you are right. probably something wrong with the last pull request. looking into it

Spiralis, 2022-02-09:
Confirmed!
Great , thx :)

closed 2022-02-09 by Spiralis