shaise/FreeCAD_FastenersWB #126: 3D printer compatibility calculations

Spiralis, 2022-02-09:
I was just doing a test with custom tap and die just now. 

For "standard" thread generation I guess it is ok (tight tolerance it seems, but that might be ok).

For "3d printer compatible" it has some settings that can be applied. 

I am wondering if the defaults are any good, actually:

[image]

If I make a custom tap and die with 30mm diameter, with a 3mm pitch, then the difference between the tap and die is ≈ 1,2 mm. 
If I make a custom tap and die with 30mm diameter, with a 3mm pitch, then the difference between the tap and die is ≈ 2,7 mm.

Now, for the largest set, 2,7 mm is almost the same as the pitch, so the threads will bearly hold at all. 

I have a feeling that the tolerance should always (?) be fixed for a bolt and nut. Meaning that Scale A should always be 1. 
Scale B is then the differentiating factor. 

I mean, the 3D printer is just as accurate for a large diameter as for a narrow diameter. The tolerance between a screw and nut is probably the same for both? That is my hypothesis at least.

Another way to approach this could be to look at this based on pitch. As in, make sure that a percentage of the pitch width is added/removed to allow for 3D printer "bleed"/inaccuracy.

In the sample below I have created a comparison between a few different methods. The first columns are the input, used for three different methods. The samples use a diameter, plus a pitch to show the effects. 

The three next column.sections are first, the current Fastener calculations (based on the module defaults). Then my changed defaults, given the hypothesis that the space should not change based on the diameter at all, and instead be based on a fixed value. Then the final example, based on the idea that the space should be a percentage of the pitch. 

I myself am leaning towards thinking that this **should** be a fixed value (example 2). 

[image]

Any thoughts on this? Are there any discussion for the basis of the current calculations that might contain information that I don't have?

shaise, 2022-02-10:
Hi,

Thanks for this elaborate report. I think it should be shared in the freecad forum, so others may contribute from their experience. 
In theory, yes, no need for the 'A' value as we want a fix gap. However, it is not only compensating for over/under extrusion, it is also compensating for plastic shrinkage after cool down. 
Furthermore, there are no 'good' default values as it depends on the printer and the filament material. The only thing was just to give an option for 3d printing compensation with basic linear manipulation of measurements. But each user must experiment with the values to fit his/hers own gear.

Spiralis, 2022-02-10:
Ah. I didn't think about shrinkage :)

But, for a case when I print two parts in 3D (screw and nut), they will both shrink with the same percentage.  If the nut is (considerably) larger than the screw, then of course this is not necessarily true.

Shrinkage is also something that should then lead to both nut and screw needing to be enlarged, right? Like, the part is always printed i.e. 0,5% smaller. So, the whole print should be scaled, not just the threads.

But, I agree that this is probably something that should be discussed in a general forum. Maybe not even just FreeCAD. Maybe even at a higher level, as this is a general 3D printing issue. 

Other CAD programs might also have a system for handling this, I guess? I have no experience with any of the other programs though. 

My "goto" solution would probably be to create a custom die and tap where I manipulate the diameter to include the tolerance. So, for steel milling the tolerance would be in 0.001 milimeters, whereas for 3D printed plastic it would be 0.1 millimeters, I guess...?

shaise, 2022-02-10:
Your guess is as good as mine.
For 3d printing I can tell you the tolerances varies widely between different printers and different filaments

luzpaz, 2022-05-03:
Please post to the freecad forum for further discussion

luzpaz, 2023-03-06:
@Spiralis please follow up

Spiralis, 2023-03-06:
I kinda solved this for my needs by making a parameter for the tolerance, and just testing my way through this, until I found a good fit. Not sure if better solutions really exist. This is a tricky topic...

closed 2023-03-06 by Spiralis