shaise/FreeCAD_FastenersWB #131: Fasterners and STEP files

ToniTen, 2022-03-22:
Crossposted from [the forum](https://forum.freecadweb.org/viewtopic.php?p=582079#p582079)

On imported STEPs, cylinders are split into two half-cylinders, and that seems to confuse this function and it ends up with duplicate fasteners. Take this example for a duct flange, that should normally be secured with 30 bolts. If you select the face and try to automatically place the bolts, you end up with not only double the amount of bolts, but also 4 extra screws place in the center.
[Fasteners and Steps.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/8323857/Fasteners.and.Steps.FCStd.zip)
[Duct_Flange.STEP.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/8323858/Duct_Flange.STEP.zip)

luzpaz, 2022-05-31:
@shaise Is this fixed by #132 ?

shaise, 2022-05-31:
I believe it should but have not tested

luzpaz, 2022-05-31:
@ToniTen ping  
Do you mind testing?

ToniTen, 2022-06-16:
> Do you mind testing?

Sorry for the late reply, but yeah, it's fixed

closed 2022-06-16 by ToniTen