shaise/FreeCAD_FastenersWB #144: Missing lengthCustom field for FSPcbSpacer

ArthurFDLR, 2022-06-25:
It would be great to be able to custom the length of Spacers similarly to StandOffs.

The `Custom` entry is available in the length parameter drop-down list, but the desired length can't be specified. There is no `Custom length` field like for standoffs. The selection of custom length leads to the following error:

```
  File "[...]\PEMInserts.py", line 1069, in execute
    l = psFindClosest(PSPMTable, PSPLengths, d, fp.width, fp.length)
  File "[...]\PEMInserts.py", line 755, in psFindClosest
    if float(len) <= float(l):
<class 'ValueError'>: could not convert string to float: 'Custom'
```
[image]

luzpaz, 2022-06-25:
Please always post your full [About](https://wiki.freecad.org/About) info (which also tells us what FastenersWB version you're using)

ArthurFDLR, 2022-06-25:
Sorry about that, here is my _About_ info:

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.0
  * fasteners 0.3.45
```

shaise, 2022-06-26:
Fixed. Please check.

ArthurFDLR, 2022-06-26:
Thanks for the quick fix. It's working.

luzpaz, 2022-06-27:
Closing ticket

closed 2022-06-27 by luzpaz