shaise/FreeCAD_FastenersWB #145: ScrewDie and

Thamone, 2022-06-26:
I have a problem with ScrewDie and ScrewTap tool. If I select diameter from the list, their size will match. But if I set diameter to the same "Custom value" the size of Die and Tap is too much off.

[image]

[image]

OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24415 (Git)
Build type: Release
Branch: makepkg
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.10.4
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

luzpaz, 2022-06-26:
CN you reproduce on the latest FreeCAD stable  v0.20? Also can you repaste About info when you do so we know what version of Fasteners WB you're using

Thamone, 2022-06-27:
Thank you, I'm building Freecad0.20 From AUR and will report you back. For the workbench version, I think it's the latest one because I've tried check addons for update but all are up to date.

shaise, 2022-06-27:
I can confirm similar results on v20.

Thamone, 2022-06-27:
I'm now on 

[code]
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.29321 (Git) AppImage
Build type: Release
Branch: master
Hash: 8ee773362e08d703bce9cfdcefd8e4ad424a7d44
Python 3.10.5, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.3.46
[/code]

I still get the same result but I now change **Thread generation** to **standard mode** and it seem to help.
[image]
[image]

I'm now set Tap offset = 0.15, Die offset = -0.15, will see if it can fit together.

Thamone, 2022-06-27:
It is printed out perfect fit with offset 0.15, -0.15 :) 
Thank you
[image]
.

closed 2022-06-27 by Thamone

Thamone, 2022-07-02:
Oh sorry my bad :P I thought this value was tolerances between the internal and external thread.