shaise/FreeCAD_FastenersWB #146: [Feature Request] Library of threaded PVC pipe adapters for folks in the US

bhs67, 2022-06-29:
I go to the hardware store and buy a PVC 1 1/2 inch male adapter. The threaded diameter is slightly smaller on the front end, so it tightens as I screw it on a 1 1/2 inch female adapter.

All PVC pipes and threaded adapters are the same throughout the United States. Whatever I buy, everything fits together.

It would be good if there were a standard library of threaded PVC pipe adapters for US folks.  No need to replicate other shapes / configurations. Just a short stub that includes threads on one end.  Ditto for female adapters. Nothing else is needed.

Forums discussion:  https://forum.freecadweb.org/viewtopic.php?f=3&t=69504&sid=c094f003e5fc2bb727d55f351b30e878&start=10

[image]

luzpaz, 2022-07-12:
Interesting models produced by the opensourceecology folks: https://wiki.opensourceecology.org/wiki/SEH_2_Kitchen_and_Bathroom_Part_Library

bhs67, 2022-07-13:
Thanks.  I was unaware of this library.

luzpaz, 2023-03-22:
@alexneufeld do you think such a feature should be available in this WB?

alexneufeld, 2023-03-22:
No. 

- I'd rather focus efforts related to this workbench on things like pins, tapping screws, and rivets. We support a **Lot** of part types already - it's probably a good idea to keep our scope of work limited to just general-purpose fasteners.
-  The OpenSourceEcology resources you linked, as well as the Flamingo workbench, already seem to do a decent job of plumbing related stuff.
- Ideally the FreeCAD will be getting [Improved Parts Library Support](https://forum.freecad.org/viewtopic.php?t=63140) in the near future. I'd rather see this sort of request implemented in that system when it becomes avbailable.