shaise/FreeCAD_FastenersWB #147: need m7 diameter

alnjmalsaat3, 2022-07-05:


luzpaz, 2022-07-05:
Please put more effort in to this feature request

FreeCAD-Tools, 2022-09-27:
As far as I know, the diameter of 7 mm (M7) cannot be specified in any way.

FreeCAD-Tools, 2022-10-03:
Standart part used according to its standard can only have a certain list of permissible diameters.

If you want an m7 screw, but there is no such diameter in the standard. Then how to build its 3d model? Head of screw, for example? To build a part, many different values are required, and they are all set out in the tables of standards. And if you do not have the diameter you need in the table, then will not possible to create the part.

However, it also happens that the csv value tables in the workbench code are incomplete or incorrectly filled in. This already requires correction.

luzpaz, 2023-03-22:
> However, it also happens that the csv value tables in the workbench code are incomplete or incorrectly filled in. This already requires correction.

@FreeCAD-Tools please cite where the inaccuracy is

hasecilu, 2023-12-28:
Related to:
- [ISO 261 norm](https://cdn.standards.iteh.ai/samples/4165/ec50482690954e82931c52fdce3bc8bb/ISO-261-1998.pdf), **ISO general purpose metric screw threads — General plan** and 
- [ISO 262 norm](https://cdn.standards.iteh.ai/samples/4167/365d2316ebbe496e87cc7e365bdc8331/ISO-262-1998.pdf), **ISO general purpose metric screw threads — Selected sizes for screws, bolts and nuts**

Most common diameters are on column 1, then in column2 are the secondly preferred, and then is a third column.

M7 is the 2nd column.

Maybe we should implement these norms as csv files to have them available for people that want to implement using this 2nd choice sizes.

In Wikipedia [thread](https://en.wikipedia.org/wiki/ISO_metric_screw_thread#Preferred_sizes) article this is marked as R10/R20 series.

hasecilu, 2023-12-29:
The new `iso626defcsv` file, aimed to replace `MetricPitchTable.csv` could be like this but the red cells highlight data that is in the Wiki article but not in the file I sent link. Should be added? Last norm version seems to be from this year, 2023.
[image]