shaise/FreeCAD_FastenersWB #159: thread to short on screws?

berberic2, 2022-08-04:
It seems, that the threads on screws are too short. 
I recognized it for ISO 14583, but might hold probably for all.
Maybe a side-effect of the fix from https://github.com/shaise/FreeCAD_FastenersWB/issues/119 ?

[image]

shaise, 2022-08-05:
This is most probably not a side effect of #119, as #119 touched only the threaded rod.
However there is definitely an issue with the thread creation. Aside from not starting from the top, adding threads seems to alter the length of the fastener. And this affects many other fasteners, if not all. I need to dig into it to see how to globally fix this issue.
Thanks for the catch.
shai

shaise, 2022-08-07:
I had to completely change the interface to the thread creation as it was too complicated to understand and use.
This should solve this issue as well.

luzpaz, 2022-08-07:
Thanks for refactoring and simplifying the code semantically!

shaise, 2022-08-10:
@berberic2, can you verify the issue is fixed?

berberic2, 2022-08-10:
Yep.
As far as I can see, it works fine now.
Thanks.

berberic2, 2022-08-10:
closing issue…

closed 2022-08-10 by berberic2