shaise/FreeCAD_FastenersWB #161: Custom Fastner

leoheck, 2022-08-15:
I am thinking there that this workbench could have a genetic way to specify a custom fastener. 

Like a button that presents a bunch of windows with some configs where we could model an unknown screw that we already have. Or where we can create a custom screw to use it on special cases. 

What do you think?

shaise, 2022-08-15:
A GUI to create custom screws could indeed be a nice feature. But I guess it will not be so easy to implement.
Right now I'm working to simplify and standardize fasteners creation in code, so maybe in the future it will be easier to add this feature.

leoheck, 2022-08-15:
Hi @shaise, yeah, no rush. Just wanted to share this idea to start a discussion. Also, the idea does not need to start and appear available ready, it can evolve with time. But thanks for the nice response. I appreciate that.

AlansCodeLog, 2023-01-24:
Would be nice to be able to at least "unlock" more properties where it makes since, like custom length does for length. I have, for example, some washers that don't correspond to the listed standard, I'm not even sure what standard some of them are. Also I often make custom thick size washers to 3d print. And the washers in the workbench don't allow any sort of customization of diameters or thickness.