shaise/FreeCAD_FastenersWB #167: Creating a thread with a custom diameter crahses the addon

nicocarbone, 2022-08-26:
I am trying to create a DIN thread (the same problem happens with a UNC thread) with a custom diameter. When I select "custom" in the diameter drop menu and try to set a diameter in the diameter input, the addon crashes. 

I am using version 0.4.3 of the workbench on freecad 0.2. This used to work in previous versions of the workbench.

Here is the output:

```
15:04:53  Traceback (most recent call last):
  File "/home/nicolas/snap/freecad/common/Mod/fasteners/./FastenersCmd.py", line 448, in execute
    s = screwMaker.createFastener(self)
  File "/home/nicolas/snap/freecad/common/Mod/fasteners/./ScrewMaker.py", line 415, in createFastener
    return self.createScrew(screwTables[fastenerAttribs.type][FUNCTION_POS], fastenerAttribs)
  File "/home/nicolas/snap/freecad/common/Mod/fasteners/./screw_maker.py", line 170, in createScrew
    screw = eval(function)
  File "<string>", line 1, in <module>
  File "/home/nicolas/snap/freecad/common/Mod/fasteners/./FsFunctions/FSmakeThreadedRod.py", line 35, in makeThreadedRod
    dia = self.getDia(ThreadType, False)
  File "/home/nicolas/snap/freecad/common/Mod/fasteners/./screw_maker.py", line 934, in getDia
    dia = FsData["DiaList"][threadstring][0]
<class 'KeyError'>: ('6.0',)
15:04:53  Recompute failed! Please check report view.
```

luzpaz, 2022-08-27:
Please post your full [about](https://wiki.freecad.org/About), TIA

nicocarbone, 2022-08-27:
Here it is:

[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 161
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * fasteners 0.4.3
[/code]

shaise, 2022-08-27:
hi @nicocarbone ,
Thanks for the catch. I should be fixed now, please test.
shai

nicocarbone, 2022-08-27:
Hi @shaise 
Thank you for looking into the problem.

Unfortunately, I updated the add-on using the FreeCad add-on manager and I am still having the bug. Should I update in a different way?

shaise, 2022-08-28:
Hi @nicocarbone ,
Sorry about that it was my mistake. I fixed the bug for Tap tool, and not for threaded rod and screw die that had the same bug. Please update again and let me know.

nicocarbone, 2022-08-28:
It seems to work now :)
Thank you @shaise!

closed 2022-08-28 by nicocarbone