shaise/FreeCAD_FastenersWB #169: “move fastener to new location” does not honor placement

berberic2, 2022-08-31:
The “move fastener to new location” function does not honor the placement of an object. It moves the Fasteners to the position the selected edge (circle) would have is placement is zero.

If I try to  “move fastener to new location”  in the attached examples it does not honor the placement of body.

[fasterners_placement.FCStd.gz](https://github.com/shaise/FreeCAD_FastenersWB/files/9462785/fasterners_placement.FCStd.gz)

shaise, 2022-08-31:
Thanks for the report. I'm on it.

shaise, 2022-08-31:
Should be fixed now

berberic2, 2022-08-31:
Now the fasteners follows the object, if placement of object is changes – which is fine,

But now “move fastener to new location” does not work at all. The function does not get activated if a fasteners and a circle-edge is selected.

closed 2022-08-31 by berberic2