shaise/FreeCAD_FastenersWB #170: “Match screw by outer thread diameter (Pass hole)” icon vanishes on switching workbenches.

berberic2, 2022-08-31:
If I open the fasterners-workbench for the first time, there are two icons, to select, if outer or inner diameter should be matched. After switching to PartDesign (or probably any other workbench) and back to fasterners the “Match screw by outer thread diameter (Pass hole)” icon is gone.
I mostly need the outer thread diameter option.

[image]

shaise, 2022-08-31:
Yes, I encountered this as well. will investigate...

shaise, 2022-08-31:
It seems to be an internal freecad bug with grouped commands. I have resided to a manual handling of the buttons. Should be OK now.

luzpaz, 2022-09-01:
@shaise is there an upstream ticket ?

shaise, 2022-09-01:
I cant say that for sure. I just did not find any reason why a button will disappear when switching workbenches

luzpaz, 2023-03-06:
Is this still relevant ?

shaise, 2023-03-06:
as the workaround works, I think we can close this

closed 2023-03-06 by luzpaz