shaise/FreeCAD_FastenersWB #180: Menu items translation bug.

FreeCAD-Tools, 2022-09-25:
At the very beginning, the normal translation is displayed. But then after a few nanoseconds it goes bad. This gif animation is made with slow motion.

[image]

shaise, 2022-09-25:
It looks line an internal freecad issue. Did you try asking in the forum?

luzpaz, 2022-09-25:
That is a weird anomaly!

riqueenz, 2022-09-25:
Apparently calling the translate function off at the beginning of the InitGui.py file solves the problem. But why?

[image]

[image]

shaise, 2022-09-26:
it looks like after the last commit, I get only partial translation when I change to Portuguese.
[image]

FreeCAD-Tools, 2022-09-26:
Tested. Works fine in not change the language.

[image]

If change the language text, labels is not changes and encodings is broke. But I think this minor bug for other issue.

[image]

riqueenz, 2022-09-26:
> it looks like after the last commit, I get only partial translation when I change to Portuguese. [image]

You need to restart FreeCAD in order to get fully translation support.

FreeCAD-Tools, 2022-09-26:
The main thing is that it works.
Close issue? Or not?

shaise, 2022-09-26:
I believe this issue can be closed

FreeCAD-Tools, 2022-09-26:
To fix that. It is necessary for freecad to cause an interrupt during the workbench, informing that the user has changed the language. But as far as I know, there are no such mechanisms now.

FreeCAD-Tools, 2022-09-26:
Ok. Closed.

0penBrain, 2022-09-28:
> To fix that. It is necessary for freecad to cause an interrupt during the workbench, informing that the user has changed the language. But as far as I know, there are no such mechanisms now.

This is a quick and IMO erroneous conclusion.
I found at least one problem : in InitGui.py, the translations use the `InitGui` context which isn't correct.
Translations in InitGui.py shall use the `Workbench` context according to documentation.
Also is should use `QT_TR_NOOP` function instead of `translate` because the `Workbench` class doesn't inherit `QObject` -- this last point don't seem to be a problem though according the .ts files of the project --.
https://wiki.freecadweb.org/Translating_an_external_workbench#In_InitGui.py

@shaise may you think this and see if it improves ?