shaise/FreeCAD_FastenersWB #183: Translation parts in treeview (Idea).

FreeCAD-Tools, 2022-09-26:
The easiest way to do this is to change the label after creating the object. For example:

        sheet = FreeCAD.ActiveDocument.addObject('Spreadsheet::Sheet',
                                                 'Fasteners_BOM')
        sheet.Label='Спецификация'

without translation:
[image]

with translation:
[image]

FreeCAD-Tools, 2022-09-26:
Good work.

Who knows where the fasteners object is created in the code?
Like (M27)x50-Screw, M16-Washer etc...
Where name of these objects is set?

[image]

shaise, 2022-09-26:
These must not be changed. it might brake files saved with fasteners

FreeCAD-Tools, 2022-09-26:
I can't figure out how the **Screw** turns into **(M14)x25-Screw**

Where does this (M14)x25 come from?

    def Activated(self):
        for selObj in FastenerBase.FSGetAttachableSelections():
            a = FreeCAD.ActiveDocument.addObject("Part::FeaturePython",
                                                 self.TypeName)
            FSScrewObject(a, self.Type, selObj)
            a.Label = a.Proxy.familyType
            FSViewProviderTree(a.ViewObject)
            print('!!!'+a.Label+self.TypeName)
        FreeCAD.ActiveDocument.recompute()
        return

FreeCAD-Tools, 2022-09-26:
Ok. This is my realization of translate. All work fine.
Files saved with fasteners is not braked...

        selfFamilyType=self.familyType
        if selfFamilyType == "Screw":
            selfFamilyType = "Винт"
        if selfFamilyType == "Washer":
            selfFamilyType = "Шайба"
        if selfFamilyType == "Nut":
            selfFamilyType = "Гайка"
        if selfFamilyType == "ThreadedRod":
            selfFamilyType = "Шпилька"
        if selfFamilyType == "PressNut":
            selfFamilyType = "Запрессовочная гайка"
        if selfFamilyType == "Standoff":
            selfFamilyType = "Стойка"
        if selfFamilyType == "Spacer":
            selfFamilyType = "Стойка дистанцирующая"
        if selfFamilyType == "Stud":
            selfFamilyType = "Шпилька запресовочная"
        if selfFamilyType == "ScrewTap":
            selfFamilyType = "Метчик"
        if selfFamilyType == "ScrewDie":
            selfFamilyType = "Плашка"
        if selfFamilyType == "Insert":
            selfFamilyType = "Втулка"
            
        dispDiam = self.CleanDecimals(self.calc_diam)
        if hasattr(fp, 'length'):
            dispLen = self.ActiveLength(fp)
            dispWidth = ""
            if hasattr(fp, 'width'):
                dispWidth = 'x' + fp.width
            label = dispDiam + dispWidth + 'x' + dispLen
            if self.leftHanded:
                label += 'LH'
            label += '-' + selfFamilyType
            fp.Label = label
        else:
            fp.Label = dispDiam + '-' + selfFamilyType

        # selfFamilyType = s[1]
        fp.Shape = s

result:

[image]

FreeCAD-Tools, 2022-09-26:
It is very strange why only screws can have the LH parameter? After all, nuts can also have it.

    if hasattr(fp, 'length'):
        dispLen = self.ActiveLength(fp)
        dispWidth = ""
        if hasattr(fp, 'width'):
            dispWidth = 'x' + fp.width
        label = dispDiam + dispWidth + 'x' + dispLen
        if self.leftHanded:
            label += 'LH'

FreeCAD-Tools, 2022-09-26:
Part workbench use translation of objects and all works fine...

[image]

FreeCAD-Tools, 2022-09-27:
@shaise and @riqueenz To make that work, need to update the *.ts files. but I don't have access to the pylupdate5 utility in windows OS.

FreeCAD-Tools, 2022-09-27:
@shaise and @riqueenz

I make code for translate parts names in treeview.
And fix LH suffix for Nuts fasteners.
(If specify that the nut has a left thread, the suffix did not appear.)
To make translation work, need to update the *.ts files. but I don't have access to the pylupdate5 utility in windows OS.

By the way, it seems to me that:

    selfFamilyType=self.familyType
    if selfFamilyType == "Screw":
        selfFamilyType = translate("FastenerCmd", "Screw")
    if selfFamilyType == "Washer":
        selfFamilyType = translate("FastenerCmd", "Washer")
    if selfFamilyType == "Nut":
        selfFamilyType = translate("FastenerCmd", "Nut")
    if selfFamilyType == "ThreadedRod":
        selfFamilyType = translate("FastenerCmd", "ThreadedRod")
    if selfFamilyType == "PressNut":
        selfFamilyType = translate("FastenerCmd", "PressNut")
    if selfFamilyType == "Standoff":
        selfFamilyType = translate("FastenerCmd", "Standoff")
    if selfFamilyType == "Spacer":
        selfFamilyType = translate("FastenerCmd", "Spacer")
    if selfFamilyType == "Stud":
        selfFamilyType = translate("FastenerCmd", "Stud")
    if selfFamilyType == "ScrewTap":
        selfFamilyType = translate("FastenerCmd", "ScrewTap")
    if selfFamilyType == "ScrewDie":
        selfFamilyType = translate("FastenerCmd", "ScrewDie")
    if selfFamilyType == "Insert":
        selfFamilyType = translate("FastenerCmd", "Insert") 

may changed to:

    selfFamilyType = translate("FastenerCmd", self.familyType) 

after generation of *.ts files...

riqueenz, 2022-09-27:
@FreeCAD-Tools I made a pull request in your own repository of fasteners WB with the updated .ts files.

shaise, 2022-09-27:
After translating selfFamilyType, 
Did you try saving fasteners on one language then open it with another language?

FreeCAD-Tools, 2022-09-27:
> After translating selfFamilyType, Did you try saving fasteners on one language then open it with another language?

I checked everything seems to be fine. I added a translation into Russian and made a commit . you can download the workbench from my repository and test it too. 

https://github.com/FreeCAD-Tools/FreeCAD_FastenersWB

For example, you can create a nut in the English language mode. save a file. switch to Russian. restart freecad, change the any parameter of fastener and its name will be translated into the language that is currently selected.

FreeCAD-Tools, 2022-09-27:
I note that I don't like that the treeview fasteners names conflict with the already existing menu names.

        translate("FastenerCmd", "Washer")
        translate("FastenerCmd", "Nut")
        translate("FastenerCmd", "ThreadedRod")

is already exist in FastenersCmd.py

Can we change keyword "FastenerCmd" to "FastenerCmdTreeView" for treeview translations?
Does it allow QT translation mechanisms?

riqueenz, 2022-09-27:
>     translate("FastenerCmdTreeView"

Yes, it works when you change it. This is called a context.

shaise, 2022-09-27:
also, I think you should use your shortcut idea:

selfFamilyType = translate("FastenerCmd", self.familyType)

riqueenz, 2022-09-27:
> also, I think you should use your shortcut idea:
> 
> selfFamilyType = translate("FastenerCmd", self.familyType)

According to the FreeCAD documentation this doesn't work.
https://wiki.freecadweb.org/Translating_an_external_workbench

> # This works:
> FreeCAD.Console.PrintMessage(translate("context", "My text") + "\n")
> 
> # This does not, lupdate only sees the word "a_variable", and doesn't know what that is:
> a_variable = "My text"
> FreeCAD.Console.PrintMessage(translate("context", a_variable ) + "\n")
> 
> # But this works -- a_variable will contain the translated string:
> a_variable = translate("context", "My text")
> FreeCAD.Console.PrintMessage(a_variable  + "\n")

FreeCAD-Tools, 2022-09-27:
> also, I think you should use your shortcut idea:
> 
> selfFamilyType = translate("FastenerCmd", self.familyType)

This idea needs to be checked.
But all the same, the start of code should have an enumeration of translation lines like:

    temp=translate("FastenerCmd", "Washer")
    temp=translate("FastenerCmd", "Nut")
    temp=translate("FastenerCmd", "ThreadedRod")

this needs for  pylupdate5 utility

FreeCAD-Tools, 2022-09-27:
Very strange, but it works for me...

       selfFamilyType = translate("FastenerCmd", self.familyType)

I add commit for tests.

https://github.com/FreeCAD-Tools/FreeCAD_FastenersWB

FreeCAD-Tools, 2022-09-27:
@riqueenz yes **lupdate** dot look this. (You use **pylupdate5**).
That's why I say that at the beginning of the code need to put meaningless strings like:

      temp=translate("FastenerCmd", "Washer")
      temp=translate("FastenerCmd", "Nut")
      temp=translate("FastenerCmd", "ThreadedRod")

This is necessary so that **pylupdate5** knows that there are such words for translation and adds them to the *.ts file.

FreeCAD-Tools, 2022-09-27:
Last test on:
https://github.com/FreeCAD-Tools/FreeCAD_FastenersWB

All workfind.It remains to check whether the *.ts files are generated correctly.

@riqueenz can you download original workbench:
https://github.com/shaise/FreeCAD_FastenersWB

replace file FastenersCmd.py to https://github.com/FreeCAD-Tools/FreeCAD_FastenersWB/blob/master/FastenersCmd.py

Remake *.ts files by you're script. And post /translations folder there in zip archive?

riqueenz, 2022-09-27:
update ts files

riqueenz, 2022-09-27:
@FreeCAD-Tools I maded a pull request at your branch

FreeCAD-Tools, 2022-09-27:
This is the end result. If everything is in order, can do a merge.

FreeCAD-Tools, 2022-09-27:
This is the final result. If everything ok, can do a merge.

closed 2022-09-28 by FreeCAD-Tools