shaise/FreeCAD_FastenersWB #188: Fasteners settings tab. Is possible to translate?

FreeCAD-Tools, 2022-09-27:
[image]

luzpaz, 2022-10-03:
@shaise  Is this a .ui file or is it generate in the code?

shaise, 2022-10-03:
It was originally generated. dont remember if it was modified manually since

FreeCAD-Tools, 2022-10-03:
https://wiki.freecadweb.org/Translating_an_external_workbench

    If you have .ui files, you need to run lupdate first:

    lupdate *.ui -ts translations/uifiles.ts

It seems to me that the ui file should just be scanned by some kind of utility like lupdate.

FreeCAD-Tools, 2022-10-03:
I run commands in windows shell:

    lupdate FSprefs.ui -ts fsprefs.ts
    lconvert -i fasteners_ru.ts fsprefs.ts -o fasteners_ru.ts

Make translate. Realase translate. All work...

[image]

But needs some kind of universal script to update all languages at once.

FreeCAD-Tools, 2022-10-04:
I think the problem is in this script:

https://github.com/shaise/FreeCAD_FastenersWB/blob/master/translations/create_ts.py

It does not find *.ui files where they stay ...