shaise/FreeCAD_FastenersWB #190: Uknown issue

leoheck, 2022-09-29:
Hey shaise, I was testing other things and see this on my terminal now

I am not sure how to reproduce it, or what is going on, I am just sharing this with you because maybe you may have a clue. 

```
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 312, in onDocumentRestored
    def onDocumentRestored(self, obj):
```

Please, close this issue if it is nothing. 

```
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.30454 (Git) AppImage
Build type: Release
Branch: master
Hash: 5df954690787dedbbf3c35f658f010f388474676
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Part-o-magic.backup1662467294.9039905
  * A2plus 0.4.59c
  * Curves 0.5.8
  * fasteners 0.4.12
  * lattice2 1.0.0
  * FreeCAD_Assembly4 0.12.4
  * kicadStepUpMod 10.15.1
  * Help 1.0.3
```

shaise, 2022-09-29:
hi @leoheck ,

Thanks for the report. There should have been some more lines describing the issue (just before or just after the line you sent), do you have it by any chance?

leoheck, 2022-09-29:
Yeh, this is not good, but I don't have anything else. I got this from the Linux terminal, and not from the inside of Freecad. It may have happened after upgrading the WB yesterday. If I see something else, I will update the issue here. Sorry, Shaie, for bringing this incomplete report.

shaise, 2022-09-29:
That's OK, Thanks. If you encounter it again please let me know.
Lot of changes were made lately

leoheck, 2022-09-29:
Thanks, dude, most of the Freecad devs do not understand that sometimes users do the best they can, even if ii is not close to what is expected. and sure, I will let you know, for sure.

luzpaz, 2022-09-30:
>  most of the Freecad devs do not understand that sometimes users do the best they can  

They understand. It's just they are inundated by requests for assistance and those usually require some education on how to report issues or request help. It can be quite a grind.

luzpaz, 2023-03-06:
Have you been able to reproduce @leoheck  ?

leoheck, 2023-03-06:
I could not reproduce this anymore, @luzpaz. Thanks for asking. I am closing this issue now. We can reopen it if something similar happens again.

closed 2023-03-06 by leoheck