shaise/FreeCAD_FastenersWB #192: Per-object tolerance settings

reivilibre, 2022-10-01:
The Fasteners workbench has some tolerance settings in edit -> preferences -> fasteners (which I think apply document-wide?), which are useful for compensating for 3D Printer inaccuracies. With a bit of trial and error, it's possible to tweak these to give some fasteners that fit well.

However I'm having the problem that the values for a well-fitting nut in the Z axis lead to a poor/loose nut in the X axis.
I'd like to be able to override the tolerances for those so that I have a workaround to be able to print my nuts/screw holes in different directions and still have decent results.

An alternative would be to have per-axis tolerance values, but this sounds a bit ridiculous and I'm not sure what you'd do if the object in question wasn't exactly axis-aligned :).

(N.B. #138 sounded both similar and different to this feature suggestion, not sure if it's a duplicate)

shaise, 2022-10-01:
hi @reivilibre 

The problem is that you can not touch the Z axis as it is changing the pitch. In my opinion there is always an xy tolerance that will give you both smooth motion and tight xy fit.

reivilibre, 2022-10-01:
You're right, my problem actually turned out to be an ID10T error :sweat_smile:.

For the model I did this time, I put an M12 screw tap on a hole of diameter 12, but for an earlier working model it was on a hole of diameter 10.2, so I wasn't actually comparing the same thing. Sorry!

closed 2022-10-01 by reivilibre