shaise/FreeCAD_FastenersWB #199: Single Button with all Fasteners

leoheck, 2022-10-03:
Not sure if this doesn't exist already. It would be nice to have a button that is a menu where we can select Fasteners from there. This way, other Workbenches could just embed a single button to access the whole workbench. 

For instance, Assembly 4 has some buttons taken from this workbench, but They are just a generic Screw, Nut, Washer and the one to change the Fastener parameters. It is missing a lot of other futures like Spaces and others types of fasteners.

shaise, 2022-10-03:
Right now the only possible grouping is a single button per group. Since there are too many fasteners, they can not be grouped into a single list, and I do not know if freecad supports sub menus under buttons.

leoheck, 2022-10-03:
Yeah, I could not find this either. But I am trying to find if this is possible, and how.

luzpaz, 2022-10-05:
What's an example of a submenu under a button ?

shaise, 2022-10-05:
Right now if you select in the preferences of fasteners that the icon grouping type is "drop down" then you will get only one toolbar button per group (instead of showing all of the fasteners). But still you as many buttons as there are groups. If you want to have a single fastener toolbar button, this button needs to drop down a sub menu of groups and each group will open a sub list of fasteners. Much like it works on the main fastener menu:
[image]

leoheck, 2022-10-05:
Exactly. A button with a dropdown list and each item with sub-items. Same behavior we have for Menus but on the Toolbar buttons.

leoheck, 2022-10-08:
Someone shared this, but I am not sure how it works

https://github.com/ebrahimraeyat/civilTools/blob/7171aacbaf3f4a0f4a27f3eac389489eaa50d1c1/gui_civiltools/define/gui_create_load_combinations.py#L124

Source: https://forum.freecadweb.org/viewtopic.php?f=10&t=72324#p630639

luzpaz, 2023-03-06:
@ebrahimraeyat can you perhaps weigh-in on this ? (it's not related to your workbenches but is  FreeCAD related :stuck_out_tongue_closed_eyes:)

ebrahimraeyat, 2023-03-08:
@luzpaz , It doesn't work for nested commands. Maybe @wwmayer can help. thanks.