shaise/FreeCAD_FastenersWB #203: Adding a bit of order to the panel (Fasteners_ToggleThreadType)

belinski-roman, 2022-10-06:
I see that the number of fasteners in your workbench is increasing and that's good. I want to propose an innovation that brings a little order to the panel. What if we add a switch button to change the two modes metric thread/inch thread (by the type of how it is done in the sketch workbench - buttons that perform Scketcher_ToggleConstruction and Scketcher_ToggleDrivingConstraint). So it would be possible to reduce and fix the number of buttons in the mode when all types of fasteners are displayed without a submenu. And if there is no corresponding analog fastener for the inch thread, then this button can be left gray.

FreeCAD-Tools, 2022-10-08:
I agree need some kind of solution. May be a panel with the ability to hide and show certain types of fasteners:

[image]

belinski-roman, 2022-10-08:
Yes, I think it's a good idea. It is necessary to think about how best to do this - in the form of a filter or as a set of checkboxes. You can also consider the option of a full-fledged sidebar instead of a workbench. This would make it possible to organize the arrangement of fastener types in a tree view of catalogs. But this is probably a topic that goes beyond the conversation about fasteners. If you implement such a thing, it is already in the context of creating a library of elements. Sincerely, Roman Belinsky.

luzpaz, 2023-03-22:
@shaise thoughts?

shaise, 2023-08-02:
Right now there is no easy way to do this, but changing the toolbar grouping to "Drop down buttons" can help quite a bit.
[image]

tobiasfalk, 2023-08-02:
What about a Dialog that pops up?
Where you then select your type, diameter and so on?
Maybe more then one Dialog, one for Fasteners, one for nut and so on.

tobiasfalk, 2023-08-02:
This is a very crude sketch of how such a wind could be structured, in this example for a fastener's for other categories like nuts or Retaining Rings it would need adjustment
[image]

berberic2, 2023-08-16:
related: https://forum.freecad.org/viewtopic.php?p=700019#p700019