shaise/FreeCAD_FastenersWB #216: [Feature Idea] Fasteners coating command.

EA32, 2022-10-18:
I propose to add a special command that could set coating parameters.

https://www.google.com/search?q=fasteners+coating&source=lnms&tbm=isch

This feature uses mainly by tehnologs.

Each metal fastener who supports coating must be able to store the following values:
1) Coating standard (ISO4042 or GOST 9.306)(List)
2) Type of coating (List or string)
3) Сoating thickness in microns(number value)
4) Coating code (read-only optional)*

*May use only the coverage code at the initial stage of creating a command.

Main purpose of this command automatically add coating code to the BOM when it generates.

In addition, after applying this command, color of the fastener may be changed.

luzpaz, 2023-03-10:
How would one go about adding and illustrating 'coating' of a fastener ?

alexneufeld, 2023-08-20:
Fastener materials and coating are out-of-scope for this workbench IMO. Or at least, this sort of information would be better captured through improvements to FreeCAD's materials system, which is planned to be improved in the near future IIRC.

luzpaz, 2023-08-20:
@davesrocketshop is working on the material system update. What do you think, Dave?

davesrocketshop, 2023-08-20:
Appearance should be possible (assuming color properties are known), but the impact on mass and other physical properties will be challenging.

luzpaz, 2025-02-10:
This should be made an upstream ticket ITC