shaise/FreeCAD_FastenersWB #217: Shape is not a shell when changing cylinder head screw parameters

james-mcjohnson, 2022-10-18:
The latest commit (e19ca7317bd6f607300dc5631d663b4d062a9b18) has the following 2 lines:

`743: res[0].Placement = FreeCAD.Placement(FreeCAD.Vector(0,0,0),FreeCAD.Rotation(0,0,0,1))
744: res[1].Placement = FreeCAD.Placement(FreeCAD.Vector(0,0,0),FreeCAD.Rotation(0,0,0,1))`

These cause a "Shape is not a shell" error when changing parameters of a cylinder head screw. By adding a reference to h_a, the error is eliminated:

`743: res[0].Placement = FreeCAD.Placement(FreeCAD.Vector(0,0,h_a),FreeCAD.Rotation(0,0,0,1))
744: res[1].Placement = FreeCAD.Placement(FreeCAD.Vector(0,0,h_a),FreeCAD.Rotation(0,0,0,1))`

shaise, 2022-10-19:
You are absolutely correct. Thanks!!

luzpaz, 2022-10-19:
@shaise JFYI, if you use `Fixes #217` then 217 will automagically be closed when commit is merged

shaise, 2022-10-19:
@luzpaz 
I prefer to let the person who opened it to check and close.

luzpaz, 2022-10-19:
Just sayin, they can always re-open if issue persists and ticket doesn't get neglected.

shaise, 2022-10-19:
Thanks. Its a good tip.

luzpaz, 2022-11-14:
Closing ticket.

closed 2022-11-14 by luzpaz