shaise/FreeCAD_FastenersWB #234: [Question]Is possible to turn and move solid body created by FSFaceMaker?

FreeCAD-Tools, 2022-10-29:
Is possible to turn and move solid body created by FSFaceMaker?
For example FSFaceMaker make contour at x,z origin. But I need create part of fastener for example at x,y origin.
Is possible create solid at x,z origin and turn it to x,y origin?
What coomands used for do that?

shaise, 2022-10-29:
Yes. Here is a small example, that turns a FaceMaker wire to the X,Y plane:
```
fm = fm = FSFaceMaker()
fm.StartPoint(0,0)
fm.AddBSpline(0,1,1,1,1,0,0,0)
w = fm.GetWire()
w.rotate(FreeCAD.Base.Vector(0,0,0), FreeCAD.Base.Vector(1,0,0), -90)
```

FreeCAD-Tools, 2022-10-30:
Yes, but this code move and rotate wire. 
But how rotate and move solid body?

      fm = FastenerBase.FSFaceMaker()
      fm.StartPoint(0,0)
      fm.AddBSpline(0,1,1,1,1,0,0,0)
      profile = fm.GetFace()
      solidbody = self.RevolveZ(profile)
      # rotate and move ...
      base_soldbody = base_soldbody.cut(soldbody)
      # or
      base_soldbody = base_soldbody.fuse(soldbody)

shaise, 2022-10-30:
it the same thing:
`solidbody.rotate(FreeCAD.Base.Vector(0,0,0), FreeCAD.Base.Vector(1,0,0), -90)`

FreeCAD-Tools, 2022-10-31:
Thanks. I think I'll use that in the near future.

closed 2023-03-06 by luzpaz