shaise/FreeCAD_FastenersWB #238: [Optimization] Notes about Settings menu

FreeCAD-Tools, 2022-10-31:
Minor notes. Perhaps this is not very important.

1. I found some values in the translation files that are not visible on the screen but taking up space in the *.ts files.
In list it marked as '?'
[image]

to that values do not includes to the *.tr file, I think they need to be marked with a special attribute notr="true"

                  <property name="prefEntry" stdset="0">
                   <string notr="true">NutThrScaleA</string>
                  </property>
                  <property name="prefPath" stdset="0">
                   <string notr="true">Mod/Fasteners</string>
                  </property>

2. Value B is looks like offset, not a scale...

If the menu will have such labels is it normal?
[image]

shaise, 2022-10-31:
regarding 2, agreed and changed. But now it should be fixed in the translations

FreeCAD-Tools, 2022-10-31:
This not easy work. What about other strings?
Now menu looks as:

[image]

And this is how it could look like:
[image]

FreeCAD-Tools, 2022-11-01:
IMO 'Thread generation' string may removed, and 'Scale settings for 3d printing' group box may be a checkable.
If the function is disabled, then the parameters are grayed out, which is clear.
This is at the discretion of the developer, and I don't know code and how to do that.

[image]

About translations.
if we change scale and offset we change totality shape of thread i.e. not only scale. Screw scaling and thread scale settings not fully correct. And formula must be more informative.

FreeCAD-Tools, 2022-11-05:
Thanks. I think the issue can be closed.

closed 2022-11-05 by FreeCAD-Tools