shaise/FreeCAD_FastenersWB #244: Notes to fasteners menu item names

FreeCAD-Tools, 2022-11-05:
https://github.com/shaise/FreeCAD_FastenersWB/blob/9d72f974b53b487a0c59258e8ff545082f02efe1/FastenersCmd.py#L102

The absence of a space between ISO and 4762, is it a typo or is it necessary for something?

https://github.com/shaise/FreeCAD_FastenersWB/blob/9d72f974b53b487a0c59258e8ff545082f02efe1/FastenersCmd.py#L203-L205

For some reason, the name and number of the standard are missing in the lines. Does that need fixing?

FreeCAD-Tools, 2022-11-05:
And who known for what is space removing added in group names?

https://github.com/shaise/FreeCAD_FastenersWB/blob/9d72f974b53b487a0c59258e8ff545082f02efe1/FastenerBase.py#L127-L134

is possibe to disable that replacing? That's looks strange:

[image]

If fix that panel name would be look as:

[image]

Corrected code:

                Gui.addCommand(subcommand, #.replace(" ", ""),
                    FSGroupCommand(cmdsubs[subcommand], subcommand, subcommand))
            else:
                cmdlist.append((subcommand, #.replace(" ", ""),
                               cmdsubs[subcommand], subcommand))

shaise, 2022-11-10:
I see the tool tip differently: 
[image]

FreeCAD-Tools, 2022-11-11:
[image]
Place the cursor over the beginning of the panel (Where the strip of dots is), after a while the tooltip should pop up. It will contain the name of the component group.

shaise, 2022-11-12:
Yes you are right.
I will check if there was a reason for that.

By the way, I usually use the drop down buttons option, it is much less cluttered.
[image]
[image]

and also the group name looks ok:
[image]

FreeCAD-Tools, 2022-11-12:
For everything to work properly, need to add one more replace at: cmdlist.append(...
It's not clear why this was added.

    def getCommands(self, group):
        cmdlist = []
        cmdsubs = {}
        for cmd in self.commands[group]:
            command, subgroup = cmd
            if subgroup is not None and GroupButtonMode > 0:
                if not (subgroup in cmdsubs):
                    cmdsubs[subgroup] = []
                    if GroupButtonMode == 2:
                        cmdlist.append(subgroup) # .replace(" ", ""))
                        cmdlist.append("Separator")
                cmdsubs[subgroup].append(command)
            else:
                cmdlist.append(command)
        for subcommand in cmdsubs:
            if GroupButtonMode == 2:
                Gui.addCommand(subcommand, #.replace(" ", ""),
                    FSGroupCommand(cmdsubs[subcommand], subcommand, subcommand))
            else:
                cmdlist.append((subcommand, #.replace(" ", ""),
                               cmdsubs[subcommand], subcommand))
        return cmdlist

luzpaz, 2023-03-06:
Is this ticket still relevant?

FreeCAD-Tools, 2023-03-06:
Partially yes.

@luzpaz You decided to close all my issues?

luzpaz, 2023-03-06:
Only coincidentally. Just cycling through the issue queue to check what's still relevant. Sometimes tickets are neglected to be closed

FreeCAD-Tools, 2023-03-07:
The main problem has been solved. I think that can be closed.

closed 2023-03-07 by FreeCAD-Tools