shaise/FreeCAD_FastenersWB #246: Some ScrewTap sizes produce bad geometry when threaded.

macdroid53, 2022-11-11:
I'm using the latest dev version 0.21 30972 and fasteners 0.4.23

[image]
[FastenersError-mac.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/9993049/FastenersError-mac.FCStd.zip)

Am I doing something wrong?

alexneufeld, 2022-11-12:
nothing wrong on your end. The thread generation code needs cleaning up. Give me some time to put a patch together.

TomSaw, 2022-12-11:
I'm also affected: diameter >= M24 does not produce solids anymore.
[image]

commit b4b21538 (2022-12-13): FSmakeScrewTap: improve reliability

resolves #246

use CreateInnerThreadCutter and simpler CSG operations to generate
correct geometry with a wider range of input parameters

This method may be slightly less performant, but the increase in
relability makes for a good trade-off
closed 2022-12-13 by shaise