shaise/FreeCAD_FastenersWB #252: Threads extending beyond nut body

zvercodebender, 2022-12-02:
I noticed recently that the threads extend beyond the nut body.  As an example I created a M10 ISO4033 nut and when I set the `thread` property to `true` the treads extend beyond the body.  Screen shot below:;

[image]

luzpaz, 2022-12-02:
Full [About](https://wiki.freecad.org/About) info please

zvercodebender, 2022-12-02:
sorry distracted....

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60e
  * fasteners 0.4.24
  * Assembly4 0.12.4
[/code]

luzpaz, 2022-12-02:
Can you provide a file for us too ?

zvercodebender, 2022-12-02:
[fasteners.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/10144818/fasteners.FCStd.zip)

shaise, 2022-12-03:
Could not reproduce the problem here:
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.24
  * sheetmetal 0.2.59
```

zvercodebender, 2022-12-03:
I did have to adjust the preferences to get the nuts and bolts to fit together.

[image]

shaise, 2022-12-03:
Yes! indeed this is the problem. 1.1 Scale is rather extreme (10% !), that's why I have not noticed that. I will see if I can fix it.

zvercodebender, 2022-12-03:
The nut scaling seemed to fix the tolerance issues on a M5 nut and bolt.  Without that the nut would not go on the bolt when printed.  I'm not sure if there might be a better way of handling that.

shaise, 2022-12-03:
Please update and test again, It should work now with any 3D print settings.

shaise, 2022-12-03:
> The nut scaling seemed to fix the tolerance issues on a M5 nut and bolt. Without that the nut would not go on the bolt when printed. I'm not sure if there might be a better way of handling that.

Having to go all the way to 10% change to make the print fit is rather extreme. usually 1.03 (3%) is good enough. Perhaps the filament thickness is not accurate, the printer flow is too high, or the layer height is too big. (for M5, 0.1 layer height is recommended)

zvercodebender, 2022-12-03:
So I changed the layer height to 0.1 mm, with the offsets at 1.05 and 0.150.  The printing was a little better, the nut almost fits on the bolt.  The model still shows the threads coming out of the nuts if the nut scaling `A` goes above 1.05 or `B` goes above 0.169.  I'm still messing around with the slicer settings to see if I can get a better print.

shaise, 2022-12-04:
A good way to test the printer/filament combination is to print a 1/2" cylinder, and, say 1" cube with 1/2" hole and using a caliper to see the difference between the hole and the cylinder. 
Anyway the fix I made lets you use the 1.1 setting without the issue.

zvercodebender, 2022-12-15:
Seems to be working now.  I also found that I needed to so some calibration.

closed 2022-12-15 by zvercodebender