shaise/FreeCAD_FastenersWB #257: [Feature reqest] Add pitch Custom support for Thread Rods and Dies.

FreeCAD-Tools, 2022-12-15:
At the moment, in the thread pitch or die settings, is possible to change the pitch Custom parameter. However, changing it has no effect. Is it possible to make this feature work or not?

Topic from forum: https://forum.freecadweb.org/viewtopic.php?f=29&t=74386

> I'm taking my first steps in 3D modeling and working in a CAD program in general. I ask for help from those who already understand the topic.
> The task is to make a cap model with an internal thread M30x1.5
> I will still deal with the object itself, because it's all for the first time. The difficulty arose immediately when trying to make a cylinder with a thread. As part of the Fasteners workbench, in addition to bolts and nuts with threads according to DIN standard, there are taps and dies (lerki), where you can (as I understand) simulate threads of different diameters and pitch. Everything is fine with the diameter, when choosing a tap, M6x1 is added, in the parameters I changed the diameter to M30. But when you change the pitch (pitch Custom), nothing changes. I put 1 or 1.5 or 3 -- on the model 3 turns at 10mm length and that's it. With the die, everything is similar, the diameter can be changed, the pitch is not.
> Tell me, what am I doing wrong? Or am I trying to use the wrong tool? Then which one is needed?

shaise, 2022-12-15:
The custom pitch does work!
But you need to change the diameter form M... to Custom.
In the past it was done automatically when touching pitch custom, So I will take a look and fix it as well.

FreeCAD-Tools, 2022-12-15:
Thanks. All works.

> So I will take a look and fix it as well.

If there is such a possibility, it would make the workbench more convenient.

shaise, 2022-12-15:
It is done now

FreeCAD-Tools, 2022-12-16:
Thanks.
Issue is closed.

closed 2022-12-16 by FreeCAD-Tools