shaise/FreeCAD_FastenersWB #263: ISO-10642 countersink angle incorrect

ChrisDumont, 2023-02-17:
From the Fasteners workbench the default ISO 10642 M6x8 hexagon socket countersunk head screw has the correct 90º countersink angle. But an M2.5x4 screw is about 78º. It should also be 90º shouldn't it?

alexneufeld, 2023-02-26:
Could you confirm that you are using the latest version of the workbench?
I can't reproduce this issue:
[image]

The countersunk head generation code was updated recently (cac2e85)

ChrisDumont, 2023-02-27:
I found the problem. I installed Freecad around January 2022 and had forgotten that the Fasteners workbench was installed separately. I've recently installed 0.20.2 but didn't remember to update the workbench. This issue can be closed. Sorry for the noise.

Thanks,
Chris

> On Feb 26, 2023, at 10:46 AM, Alex Neufeld ***@***.***> wrote:
> 
> 
> Could you confirm that you are using the latest version of the workbench?
> I can't reproduce this issue:
> 
> 
> The countersunk head generation code was updated recently (cac2e85 <https://github.com/shaise/FreeCAD_FastenersWB/commit/cac2e851475cf098cd76ef2d99156ba067ed74e8>)
>

closed 2023-02-27 by luzpaz