shaise/FreeCAD_FastenersWB #266: Decouple heads from threads?

BloodyRain2k, 2023-04-02:
I'm right now looking through the WB, trying to understand how I add a "new" type of screw, which is in reality nothing more than combining the head of the default flat PAN head screw with the thread of the "Wood screws".

It's a type I have sitting on my desk right now but no idea what's named, and looking through the code, it seems that I have to copy the code from first half of the `makePanHeadScrew()` and the bottom half of `makeDIN571`.

And I just wonder, why were these parts coupled in the first place?
Wouldn't it have been easier to make one generator for heads and another for threads and the "make screw" functions just combine the results?

mwoehlke, 2023-09-09:
This would be monstrously helpful. For example, support for ASME/UTS fasteners is abysmal... for no good reason. (It's outright shocking that there is no Phillips pan head #6-32, which is the single most ubiquitous screw in computer hardware.)

If we could simply slap a UTS thread on an existing head, that would go a long, long way to remedying that problem, as well as making the library much more navigable. Besides, heads and threads *are* more or less orthogonal.

hasecilu, 2024-01-05:
The main obstacle to implement new fasteners is not the geometries but the data from particular standards.

As you noticed the heads and threads/shanks are merged because doing it in this way let us use less operations otherwise we would need the head, the body and then to join them another ring-like structure for the fillet, as of now we just need to revolve around Z half-face.

Each standard specifies its dimensions and having generic heads around would not be helpful with the current approach.

What is implemented separated are the recesses (hex, cross, slot)  and "real" threads and that's handy.

[image]