shaise/FreeCAD_FastenersWB #269: Can you provide the size pictures of the fasteners

qunat, 2023-07-29:
I want  the size pictures of the fasteners。

luzpaz, 2023-08-08:
Please make a mockup of what you are asking for.

qunat, 2023-08-08:
Thank you for replying to me .you can login www.pythonocc.com .This web is the backend program I created based on your program.l want the picture can.show those parameters.

luzpaz, 2023-08-08:
@shaise what are your thoughts ?

shaise, 2023-08-09:
@luzpaz I must say I did not quite understand what exactly is the request

luzpaz, 2023-08-09:
I think I understand but lets just make it less of  a guessing game:  
@qunat to make it easier for us...can you make a mockup of FreeCAD that shows what exactly your are proposing ?

alexneufeld, 2024-02-25:
This should be closed. Schematics of all common fastener types are freely available on the internet for OP to use in whatever custom web UI they are apparently making. Otherwise they can use TechDraw or one of the many measurement tools available in FreeCAD's GUI to make dimensioned drawings themselves

shaise, 2024-02-25:
I still did not understand the request. If it is similar to other cad sw where you can browse say mcmaster-carr catalog drawings, its a legit request, though tedious to implement.