shaise/FreeCAD_FastenersWB #272: Feature request: Thread pitch for bolts

OH1KK, 2023-08-16:
Please consider to add Thread pitch parameter on Screw parameter list. Tables which explain what coarse, fine I, fine II, fine III and fine IV thread pitches means is available on boltsparts github page https://boltsparts.github.io/en/parts/names/hexagonHeadScrew1.html

OH1KK, 2023-08-16:
Workaround before these fine pitches are implemented. For example to get DIN933 M27 with 1.5mm pitch. Open freecad/common/Mod/fasteners/FsData/din933head.csv file. Then Find M27 bolt 

`"M27",3.0,0.4,38.0,45.2,17.0,1.0,41.0`

Second value it pitch. Modify it

`"M27",1.5,0.4,38.0,45.2,17.0,1.0,41.0`

Save file. Restart freecad. Now make M27 bolt and it has 1.5mm thread pitch.

tobiasfalk, 2023-08-16:
I have looked up the DIN 933 and the ISO 4017, the replacement for DIN 933, and in neither mentions a fine pitch version of the screw. I also wasn't able to find a screw on an online screw shop with such a pitch. One possibility would be to make 4 "new" screws where the pitch is modified, it is named probably, "DIN 933 fine I, none Standard" or similar and a note is given in the description part, with a link to where these pitches come from (source).
If you have a source for where such screws are coming from (an online shop) maybe give a link.

OH1KK, 2023-08-17:
You a likely right that those are not standard, since those bolts are hard to find from any stores. But those bolts exist and they are used, for example on a Ural motorcycle fork. It's just a cap keeping water out of fork. 

https://www.ural-zentrale.de/en/chassis/telescopic-fork-dnepr/telescopic-forks-top-bolt-27-mm

It would nice to be able to 3D print spares when original bolts are lost.

Same thread is also used in and in some blumber pipe things but with a different head

https://www.rejko.fi/tuote/putket-osat-ja-liittimet/sinkityt-kierreosat/tulppa-sinkitty-uk/tulppa-sinkitty-1/

berberic2, 2023-09-05:
> I have looked up the DIN 933 and the ISO 4017, the replacement for DIN 933, and in neither mentions a fine pitch version of the screw. I also wasn't able to find a screw on an online screw shop with such a pitch. 

That's because fine thread screws have their own standards. In this case:
· DIN 961 (withdrawn and replaced by) 
· ISO 8676 / DIN EN ISO 8676

tobiasfalk, 2023-09-05:
In #285 I added DOM 961 and ISO 8676.
@OH1KK I hope those are the screws that you needed

hasecilu, 2024-01-05:
I have just added ISO 262 which is a subset from ISO 261 where the different diameters have associated to them a coarse thread and some fine threads (https://github.com/shaise/FreeCAD_FastenersWB/pull/314/commits/80615b73ae3dd86c2830e70fbabb60139c9d204b)
ISO 262 is for bolts, screws, studs, nuts, these are the recommended ones. Apart from this table I have seen bolts commercially with fine pitches that are not in ISO 262 but are in ISO 261, maybe we need to also add it.
```python
P = FsData["ISO262def"][fa.diameter][1] # fine thread 1
```