shaise/FreeCAD_FastenersWB #281: [Question] configuration files for translation handling

kaktusus, 2023-09-03:
There is nothing in the translator's [guide](https://github.com/shaise/FreeCAD_FastenersWB/blob/master/TRANSLATIONS.md) about the `create_ts.py` file.
Should it be updated when new languages are added?

shaise, 2023-09-05:
@riqueenz can you comment ?

riqueenz, 2023-09-05:
>There is nothing in the translator's [guide](https://github.com/shaise/FreeCAD_FastenersWB/blob/master/TRANSLATIONS.md) about the create_ts.py file.
>Should it be updated when new languages are added?

No, the create_ts.py and compile_ts.py files are deprecated and it is only necessary to update the update.sh file as it is in the guide.

kaktusus, 2023-09-05:
Thank you for the clarification.
:wink:

closed 2023-09-05 by kaktusus