shaise/FreeCAD_FastenersWB #283: Thread length "b" incorrect interpretation

juddidier, 2023-09-05:
Hi,
I wanted to add a new screw an saw that the "b"-parameter (thread length) is incorrectly interpreted in the code. I work only on the metric system, don't know if it also apply to imperial.

As of now, "b" is interpreted as the maximum thread length for a given screw diameter. It results that e.g. a screw ISO7048 with a length longer than 38mm always has the thread restricted to 38mm.

But The DIN and ISO norm always write "b min"(!), meaning: If the thread and shaft length aren't the same, the thread length has to be at least (for the example above) 38mm.
for screws according to ISO7048, e.g. M4:
-> screws up to 35mm length, thread goes always up to the head.
-> screws 40mm and more, thread can be smaller (min 38mm in length) but must not
At work we use such screws with length of 40-60mm where the thread goes up to the head. They are not very common (you probably won't find them in the diy-shop), but still comply to the norm.

This apply to all DIN and ISO screw (I checked several screw types and they all state the same).

It would mean to add at least a boolean parameter (in the property pan) "fully threaded". if false, either restrict the thread length to "b" or add a second parameter with the desired thread length.

Can you correct this issue?

alexneufeld, 2023-12-25:
The interpretation is not incorrect. The standards (DIN and ISO) specify that **at least** a length b of thread is required. So manufacturers are free to thread this amount or more, and designers should **not** assume that more than that length will be threaded. Otherwise, the following situation is possible:

- Someone designs a part that requires more than the minimum thread length b on some screw.
- They check that the screws in stock in their workshop have this amount of thread.
- Later, the purchasing department at the same company has to switch screw suppliers for some reason.
- The new supplier of screws doesn't supply more threads than are required by the standard.
- The original design now has to be reworked or changed, even though all fasteners purchased before and after changing supplier conform to the relevant standards.

this is sort of a 'defensive driving' thing. Designers shouldn't assume that their suppliers will do more work than needed, and we try to help with that by showing a 'worst case scenario' version of standard fasteners.

juddidier, 2023-12-31:
I have to disagree a bit. At work we often use both variants of screws: the ones fully threaded and ones only with "b" length. The suppliers have different order codes for both variants of the screws. So there is never a misinterpretation.
With your "interpretation" of the norm, a fully threaded screw is not possible.
I asked back at our main supplier (distributor of wurth): they stated that the norm allows you to have either fully threaded or partially threaded and if second, the thread must have b length - The "min" meaning a tolerance -0 / +xxx µm, +xxx µm for the production tolerance.

alexneufeld, 2023-12-31:
ISO 7048 just states "b (min)" with no commentary or description of what manufacturers *should* do as far as I can tell. 
[image]
My biggest concern is just wanting to avoid additional complexity in the fasteners objects. We already do too much IMO (such as the customizable screw overall length, left handed, real thread, and diameter matching properties). It's a mess, and FreeCAD itself makes it hard to test the behavior of and maintain backwards compatibility of scripted objects.

shaise, 2023-12-31:
I tend to agree with @alexneufeld , there are already far too much parameters per each fastener.

juddidier, 2024-01-01:
@alexneufeld I can absolutely understand your concern. and I also agree that the screws have quite enough parameters. But in my opinion I would rather dismiss the "left handed" parameter, which is extremely seldom used, rather than a quite common parameter. E.g. the Wurth catalog gives back 32 parts for "left-handed thread" (only hex-head screw(!), a few rods and nuts, and some specialized things) in opposite to "tread up to the head" giving 551 products (with >20'000 implementations)!

Sorry, but after having checked this, IMHO I don't see the left-handed parameter as valid for every screw. The parts that are valid for left-handed should be placed in a separate category - with probable (few) compatibilty issues...

To further trimm the parameter IMHO the parameter "real thread" could be made global and switched on/off in the preferences. Usually one want to hide all threads (for better performance) or show all (for e.g. presentation purpose).

tobiasfalk, 2024-01-02:
If I remember correctly, every ISO screw and bolt norm describes how a screw or bolt should be named.
I only looked up ISO 8676, but there the following example is given:
M12x1,5 l= 80mm "Festigkeirklasse" 8.8 is a **ISO 8676 -M12 x 1,5 x 80 - 8.8**
If b where a parameter that is variable like the length or the diameter, why is it not represent in the naming?
I understand that there are screws and bolts that are sold with a bigger b, but that these are in a way similar to the ISO 1207 M1.4 bolt(#267), where they comply with the norm but the size it self is not specified in the Norm.

juddidier, 2024-01-02:
@tobiasfalk you are right. I just checked the wurth catalog. They differentiate the two thread length with eg. M5x50 (for the short thread) and M5x50/47 (thread to the head). If it is part of the norm, I don't know.

tobiasfalk, 2024-01-02:
@juddidier could you do two things for me?
1st, can you ask your main supplier if he understands the "norm" name of the Fasterner? For example the one I have given before or for your example it should be something on the line of **ISO 7048 - M5 x 50 - 4.8** (I am not sure on this one because I do not have the norm to hand)
2nd, could you look up other suppliers and ask them how they designate this M5x50/47 screw

tobiasfalk, 2024-01-02:
@juddidier could it be that there is a companion standard for ISO 7048, where the thread goes up to the had and so has no shank. this is the case with hex bolts, there ISO 4014 has a shank and ISO 4017 has no shank.