shaise/FreeCAD_FastenersWB #286: List of all ISO hex bolts and nuts + which are implemented

tobiasfalk, 2023-09-05:
While looking up ISO 8676 I saw this list in the foreword of the norm and whanted to share the list so that it can be used as a guide for what is still to be implemented.

### Hex Bolts with shank:

- ISO 4014 **DONE**
- ISO 4015
- ISO 8765

### Hex Head Bolts threaded to the Head:

- ISO 4017 **DONE**
- ISO 4018
- ISO 8676 #285

### Hex Head Bolt with flange:

- ISO 4162
- ISO 15071
- ISO 15072

### Hex Nut:

- ISO 4032 **DONE**
- ISO 4033 **DONE**
- ISO 4034
- ISO 4035 **DONE**
- ISO 4036 **DONE**
- ISO 7040
- ISO 7041
- ISO 7042
- ISO 7719
- ISO 7720
- ISO 8673
- ISO 8674
- ISO 8675
- ISO 10511
- ISO 10512
- ISO 10513

### Hex Nut with flange:

- ISO 4161
- ISO 7043
- ISO 7044
- ISO 10663
- ISO 12125
- ISO 12126
- ISO 21670

hasecilu, 2024-01-02:
Good list, I think we can say if we are working on some fasteners.
Hi, I'm working on  ISO 7434/5/6.

hasecilu, 2024-01-03:
Notes: 
- ISO 299 is implemented as an alias of DIN 508, exactly the same, and the data for DIN508 is from a supplier that allows multiple hole sizes with a single slot width.
- ISO 7049, #314 
- ISO 7093-2, If the data it's the same could be implemented as an alias, no dimensional changes, only hardness, similarly, in the code ISO 7091 is not available because it's commented out due to a similarity in data to ISO 7089
- ISO 8738 #312

alexneufeld, 2024-02-19:
@tobiasfalk can you try to keep this up to date as we add more fasteners? No worries if you don't have time. 

A list of things to check off:
- I *think* all ISO pins are implemented
- ISO 7379 is done
- ISO 7047 is done
- ISO 14583 is done
- ISO 2342 is WIP in #342 
- all remaining nuts are WIP in #343

tobiasfalk, 2024-02-19:
Yes have/had them on my radar, but did not finde the time to update the list

alexneufeld, 2024-02-20:
#344 adds:
- ISO 4015 reduced shank hex head screws
- ISO 4016 hex head bolts, grade c
- ISO 4018 hex head screws, grade c
- ISO 4162 flanged hex head bolts
- ISO 15071 flanged hex head bolts
- ISO 15072 flanged hex head bolts with fine threads