shaise/FreeCAD_FastenersWB #296: Failed cutting operation between a thread and a cilindre

Nosnahc, 2023-09-14:
Hello
I'm trying to create a thread on a cylinder using the Fastener plugin. 
I generated a ScrewDie 
[image]

and I try to make a "cut" operation but the result is unexpected.
[image]

FreeCAD : 0.21.1
OS : Linux Mint 21.2
Fastener version : 0.4.66

Nosnahc, 2023-09-15:
I submit it to the [Freecad Github](https://github.com/FreeCAD/FreeCAD/issues/10682) but seams like the problem is from Fasterner side.

luzpaz, 2023-09-15:
Please add example file (zip it and drag-drop it into github)

Nosnahc, 2023-09-15:
Let me know if you need something else.
Thanks by advance!
[Fastener_Probem.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/12622025/Fastener_Probem.zip)