shaise/FreeCAD_FastenersWB #297: FastenersCmd.FSViewProviderTree not JSON serializable

berberic2, 2023-10-14:
I’m getting python error-messages from fasteners. Not sure why this showing now and if the problem is in fasteners or my system.
```
PropertyPythonObject::toString(): failed for <class 'FastenersCmd.FSViewProviderTree'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable

```

shaise, 2023-10-15:
Hi. How to replicate it, and what FC version?

berberic2, 2023-10-15:
Create a document with a fasteners object and save it. Version is from git.

```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34663 +2 (Git)
Build type: Release
Branch: Meins
Hash: 9b653657ff3d91e9c1fa74be960efbdc3bb31ca6
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.13
  * Assembly4 0.50.4
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.3.1
  * Silk 0.1.5
  * CurvedShapes 1.0.4
  * DynamicData 2.53.0
  * FreeCAD_assembly3 0.11.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * A2plus 0.4.60n
  * Render 2023.8.18
  * fasteners 0.4.66

```

shaise, 2023-10-15:
I do not have access to a Linux machine. I used the latest weekly build for windows and it seems to work fine. However this build uses Python 3.10 and yours is 3.11. perhaps this is the issue?

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34651 (Git)
Build type: Release
Branch: main
Hash: aa5bea4e2e8089bf00f61075a0860dacda42dc8b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67
  * sheetmetal 0.3.1
```

berberic2, 2023-10-15:
The json implementation from 3.11 is nearly identical to the 3.10 one. So, I doubt this is the cause.
```
diff -Naur /usr/lib/python3.10/json/encoder.py /usr/lib/python3.11/json/encoder.py
--- /usr/lib/python3.10/json/encoder.py 2022-12-07 14:47:07.000000000 +0100
+++ /usr/lib/python3.11/json/encoder.py 2023-03-13 13:18:29.000000000 +0100
@@ -30,6 +30,7 @@
 for i in range(0x20):
     ESCAPE_DCT.setdefault(chr(i), '\\u{0:04x}'.format(i))
     #ESCAPE_DCT.setdefault(chr(i), '\\u%04x' % (i,))
+del i
 
 INFINITY = float('inf')
```

shaise, 2023-10-15:
and this happens with any fastener object?

berberic2, 2023-10-15:
As far as I can tell: Yes.

berberic2, 2023-10-15:
Same problem on:
```
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Release
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.4.67

```

shaise, 2023-10-15:
sorry, can not reproduce it.
Do you know from which build this problem started? perhaps it is possible to see the changes that might have caused it.

berberic2, 2023-10-19:
I do not have the problem with commit 92b26354ba aka 0.21rc1 from 2023-07-31.
I do have the problem with commit aa82948b27 from 2023-10-06

berberic2, 2023-10-20:
Problem is introduced by commit [83d4080fe8](https://github.com/FreeCAD/FreeCAD/commit/83d4080fe8) from 2023-09-24
```
App: fixes #10460: App::PropertyPythonObject is not saving data
    
    Since Py3.11 the methods names __setstate__ and __getstate__ conflict with the method names added to the object class.
    Thus rename them to 'loads' and 'dumps'

```

shaise, 2023-10-21:
Nice catch!
But if Fasteners are not using any of these methods, then then theoretically it should work with the built in fix, no?

berberic2, 2023-10-21:
My understanding is, that these functions are called by freecad to save/load the state of an object.

fra589, 2023-10-31:
Hi @shaise, Hi @berberic2,  

Same problem with my FreeCAD version

Since Py3.11 the methods names __setstate__ and __getstate__ conflict with the method names added to the object class.

In [FreeCAD App: Fixe #10460](https://github.com/FreeCAD/FreeCAD/commit/83d4080fe8) they have been renamed to 'loads' and 'dumps' 
This functions are called by the FreeCAD file [src/App/PropertyPythonObject.cpp](https://github.com/FreeCAD/FreeCAD/blob/main/src/App/PropertyPythonObject.cpp)

I made the correction at home and it work for me... No more error when saving FreeCAD with Fastenersobjects.
So I made the pull request [#302](https://github.com/shaise/FreeCAD_FastenersWB/pull/302)

@++;
Gauthier.

shaise, 2023-10-31:
hi @fra589 ,

Thank for the fix, however this works only on python 3.11 and not on prev ones - hence it fails on the release version of FreeCad (0.21 for now)

fra589, 2023-10-31:
Hi @shaise,  
  
The link on the Python version is due to a conflict on the \_\_setstate\_\_() and \_\_getstate\_\_() functions names which are used internally by Python. So, wwmayer changed the name of ViewProvider's called functions in FreeCAD 0.22.

I updated the pull request #302 to to make it work for all releases of FreeCAD (tested on 0.20 (sorry, I dont have workable 0.21 now) and 0.22).
It also work with Python 3.10 and 3.11

@++;
Gauthier.

shaise, 2023-11-01:
Hi @fra589 ,

Thanks for the fix! I have merged it. Also did small cleanup + support for linkstage branch many use.

@berberic2 ,
please test if it solves your problem.

shai

berberic2, 2023-11-01:
> @berberic2 , please test if it solves your problem.

Yes. No error-messages on saving anymore.

Thanks @shaise and @fra589

luzpaz, 2023-11-01:
@berberic2 issue can be closed ?

berberic2, 2023-11-01:
Yep.

closed 2023-11-01 by berberic2

Zolko-123, 2024-01-02:
> The link on the Python version is due to a conflict on the __setstate__() and __getstate__() functions names which are used internally by Python

do you have a link explaining all this ? Because from what I read, `__setstate__()` and `__getstate__()` are still [available in Python 3.12](https://docs.python.org/3/library/pickle.html#object.__getstate__)

fra589, 2024-01-03:
Hi @Zolko-123 

Since Python 3.11, the pickle module implement the `__setstate__()` and `__getstate__()` functions.
Search for it in this link:
https://docs.python.org/3/library/pickle.html

@++;
Gauthier.