shaise/FreeCAD_FastenersWB #298: The die tool has the wrong profile. It creates the screw with the profile of a nut.

cnuehll, 2023-10-14:
green is a M6 screw and red is the ScrewDie-tool
The die-tool (red) does not match the screw (green)

I expect that the ScrewDie-tool (red) creates a screw profile matching the screw (green)

[image]

[image]

here it is well explained how the boolean between a cylinder and the ScrewDie-tool creates a screw:
https://github.com/shaise/FreeCAD_FastenersWB/issues/64#issuecomment-632810462

shaise, 2023-10-15:
You are correct. It looks like someone messed the original code which was correct. I will take a look.

shaise, 2023-10-15:
It should be fixed now. Please check.

[image]

cnuehll, 2023-10-15:
V0.4.67 works for me. I tested on freecad 0.21.1.33694  and  0.22.0dev.33909 AppImage. Thank you very much for the super fast fix.

closed 2023-10-15 by cnuehll