shaise/FreeCAD_FastenersWB #300: M3 heatset inserts as an option?

kozross, 2023-10-29:
A lot of 3D printed designs use these, and being able to put them into an assembly without having to find a model would help both designers and users. A [parametric model exists](https://print.grabcad.com/library/m3-heat-set-screw-inserts-for-3d-printing-1) for this kind of insert (SolidWorks only, unfortunately).

e-dreyer, 2023-10-29:
This is a great idea. OpenSCAD did well wit their general use libraries. FreeCAD should also support such things!

kozross, 2023-10-29:
In particular, these allow control of not only the thread (which is already possible in the existing workbench), but also the length and diameter of the insert itself, which can vary considerably. It is this variability that I'd like to be made available.

shaise, 2023-11-02:
I have added option co change length / external diameter:
[image]
[image]

please test.

shaise, 2023-11-05:
@kozross , can this issue be closed?

hasecilu, 2023-12-10:
I tested it and proves to be useful, previously I found differences in dimensions with the inserts I have and this fixes that.

kozross, 2023-12-10:
Likewise. Works great!

shaise, 2023-12-11:
good. closing this issue.

closed 2023-12-11 by shaise