shaise/FreeCAD_FastenersWB #301: Query parameters of standard fasteners via aliases

kozross, 2023-10-29:
For some context: my standard workflow in FreeCAD involves creating a spreadsheet, which I populate with values that I alias to useful names. This allows me to use them in multiple places, increases the clarity of my sketches, and allows easy re-parameterization or changes. Since I design mostly for 3D printing, a lot of these values involve things like screw diameters, cap depths, lengths of other parts, etc.

While this workbench gives me the ability to easily create, for example, DIN912 screws, I can't easily create (or get some other way) an alias for something like its cap diameter without manually measuring and hardcoding. This is suboptimal, as I'm effectively having to look up and enter values I technically already have available. Would it be possible to have some aliases for various standard measurements for standard fasteners that could be used like spreadsheet aliases currently can be?

As an example of what I'd like, consider [DIN912 screws](https://www.fasteners.eu/standards/DIN/912/). I'd like to have aliases such as `din912_m3_dk_max`, which would be 3.0mm, available if I have the Fasteners workbench installed.

shaise, 2023-11-02:
There are just too much fasteners and property combinations to create an alias for each. It will just bloat the memory.

kozross, 2023-11-02:
I've not dealt with FreeCAD's internals, but how bad could this be really?

shaise, 2023-11-04:
Just think of the auto complete feature when you enter a value in the function editor, when the list of available aliases is close to 10000.