shaise/FreeCAD_FastenersWB #304: [Feature request] Add global "enabling/disabling" real thread calculation

fra589, 2023-11-15:
Hi,

When we have many number of Fastener object in a FreeCAD model, calculation time dramatically increase at each simple operation...

Perhaps a "global" option for the "thread" parameter could save a lot of time :

[image]

Then, in the main Fastener menus, it will be an option "global threads computation" which could be "on" or "off"...
So, you can construct those complex items with short "no reals threads computing", then, when the model is finished, turn on calculation with the global option, then  save the model and export reals geometries.

What do you think about it ?

@++;
Gauthier.

alexneufeld, 2023-12-25:
This doesn't align with how FreeCAD works. A file should have the same geometry regardless of the settings on a particular users system. We already contravene this principle for the 3D printing thread geometry settings, and it's generally bad.

Instead, you should add a property to your assembly called something like 'LevelOfDetail' and use expressions to change the 'thread' property of fasteners accordingly.

An example is attached:

[image]
[level_of_detail_example.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/13767120/level_of_detail_example.zip)

tobiasfalk, 2023-12-26:
This Assembly option could be created as a standard option when this Workbench is included as am Integrated Workbench([#307](https://github.com/shaise/FreeCAD_FastenersWB/issues/307))

fra589, 2023-12-27:
Hello everyone,

I do not pretend to say what FreeCAD's philosophy should be. Other modules work with global options without this bothering anyone.

However, I agree that a file must have the same definition regardless of the system on which it is read. But the rendering of this definition can also be adapted according to the needs and performance of the target system. In this context, the same definition does not necessarily mean exactly the same geometry.

The options of another module (such as assembly) have nothing to do with the need I described. I don't use the assembly module which I consider useless as it stands.

To clarify the need, let's take for example a part on which there are several (or even several dozen of) threaded holes... With each modification, all these holes are recomputed, which takes very lot of time. On my workstation (dual Xeon + 16GB RAM), some of my models take several minutes for each operation!

Currently, I'm forced to turn off the "thread: true/false" option for each thread when I want to change the model geometry, and then turn them all back on, one by one, to do the final geometry rendering I need. Again, it takes a lot of time!!!

In my opinion, a global thread processing option is not necessarily linked to the workstation... It can just as easily be linked to the file being processed.

@++;
Gauthier.

alexneufeld, 2023-12-28:
'global' variables at the **per file** level are indeed a good feature, used by several other CAD software. Unfortunately, this functionality isn't currently implementable using FreeCAD's python API. FreeCAD documents currently have some parameters that aren't attached to particular objects, but the set of them is fixed and not accessible from expressions. There's just no way for us to store information in a FreeCAD file that isn't attached to a specific geometric object.

[image]
There has been some discussion to change this behavior recently I think, but I now can't find the relevant github issues.

fra589, 2023-12-29:
Hi @alexneufeld,  
  
Even if this does not yet exist in FreeCAD, I think it must be possible to implement in the Fastener module itself, since it's a Fastener option.
The option can be added to all Fastener objects, and in case of a change of value, the new value will be changed in all the concerned objects... So each Fastener object will know the global value.  
  
It will still be possible to move the option to the general file level when it will be implemented by FreeCAD.  
  
@++;
Gauthier.

fra589, 2023-12-31:
Hi @shaise,  
  
It is precisely because I design real parts with many threaded holes for the purpose of 3D printing that I think this functionality is required.

In the design process, we are always required to make changes to the geometry even though we have already added the threaded holes. In this case, threads are all recalculated at the same time. It takes a huge amount of time for each action in the model.

With the global processing option, the calculation of the reals threads would only be carried out once at the end, when exporting for printing. Which would save significant time!  
  
@++;  
Gauthier.

alexneufeld, 2023-12-31:
That's fine, but I still don't think that it should be implemented *by the Fasteners Workbench*. The example I previously posted solves this problem, and another popular way to go about it is to use [Spreadsheet parameters](https://wiki.freecad.org/Manual:Using_spreadsheets) (no assembly required). We don't need to re-implement functionality that simple expressions already provide.

shaise, 2023-12-31:
> We don't need to re-implement functionality that simple expressions already provide.  
But still you need to go one by one on each and every fastener, and change the 'tread' property to the new function.

fra589, 2024-01-02:
Hi @shaise, hi @alexneufeld,  
  
1st, I wish you an Happy new year! :-)

The use of [Spreadsheet](https://wiki.freecad.org/Manual:Using_spreadsheets) can be a good solution, but I don't see how to use it. The 'tread' property does not seam to be linked to the [expression](https://wiki.freecad.org/Expressions) engine and doesn't have the button to do this !?!?!?

Can you explain me how to link the 'tread' property to a spreadsheet true/false cell?

@++;
Gauthier.

cfunseth, 2024-02-13:
@fra589 In short you need to assign an alias on a cell, say for instance we call it "thread". Then you can right click on the thread property for the screws in your document and select "Expression..." where you can enter the text "Spreadsheet.thread". Back in the spreadsheet you can change the "thread" cell value to either 0 or 1, which should update all of the screws simultaneously.

Alternatively, you could also accomplish this using a macro/Python script to iterate through the screws in your file and update the "thread" property. Here's a simple script I wrote that you could use whenever you want to toggle the threads:
```
 for screw in App.ActiveDocument.Objects:
	if 'Screw' in screw.Name:
		screw.thread = not screw.thread
```

luzpaz, 2025-02-10:
What's the verdict here ?

fra589, 2025-02-12:
Hi @luzpaz, 
 
> What's the verdict here ?

The "Tread" property doesn't seam to be linked to the expression engine since it doesn't have the expression icon on it. But it's possible to add an expression by using the context menu (right click on it) and select "Expression...".
Then, in expression dialogue, it's possible to use any bollean variable (from spreadsheet, other Fastener object or any other boolean variable location).
This is not a easy as what I asked, but it work for a global "enable/disable" of all threads in a model.
So the Fastener development team thinks that it is not useful to add this global option and will not do it.
However, I think that it could be useful to add the "Expression" icon on the "Thread" option field.

@++;
Gauthier.