shaise/FreeCAD_FastenersWB #305: Hide ISO or ASME parts

wasteoinc, 2023-11-17:
As the library becomes more rich it also becomes more clogged and difficult to navigate. Is there already or could I help out to make a button that hides/shows all the ISO or ASME parts?

I live in Europe so I practically never use ASME fasteners. If you could guide me where to I could also try to implement it myself

Thanks for a great workbench

shaise, 2023-11-17:
yes indeed!
activate the workbench, then go to edit -> preferences  and select fasteners on the left.
You can then have an option to select which type of fasteners to show:
[image]

you can also change the grouping to drop-down, so it will take less space on the toolbar.

wasteoinc, 2023-11-17:
Wow!  A new world opened to me :) Thanks a bunch

closed 2023-11-17 by wasteoinc