shaise/FreeCAD_FastenersWB #307: Changing the workbench license

prokoudine, 2023-12-05:
The Workbench Creation guide [recommends](https://wiki.freecad.org/Workbench_creation) licensing addons under the terms of LGPL2.1 — the same license that FreeCAD source code is available under. The recommendation’s rationale is that it should be possible to ship a FreeCAD distribution, either upstream, or a custom one, with 3rd-party addons that are considered useful.  While a workbench developer is free to choose an alternative license, doing so may impact the long-term viability and use of the workbench and should be considered carefully.

Ondsel would like to integrate the Fasteners workbench into its distribution of FreeCAD, because the company’s market study suggests that this feature is expected to be available in 3D CAD programs by default. We want to provide our end-users the best possible experience. There’s a possibility that this workbench then will go back to become part of upstream FreeCAD. However neither the upstream version of FreeCAD, nor Ondsel’s version of FreeCAD can distribute this workbench as long as it’s licensed under the terms of GPL2.

We are asking for a conversation about relicensing this workbench under LGPL2+ or another compatible license to allow wider distribution.

alexneufeld, 2023-12-05:
I have no issues with the idea of doing this. Taking a quick look through the repository contributors list, I'd say there's a good chance we can get everyone to sign off on a license change to LGPL. I agree that this change fits well within the philosophy of the FreeCAD project. 

For potentially useful reference: I like the approach the MPV player developers took to tackling this exact problem: [https://github.com/mpv-player/mpv/issues/2033](https://github.com/mpv-player/mpv/issues/2033)

prokoudine, 2023-12-05:
@alexneufeld That's a really useful reference, thank you!

shaise, 2023-12-05:
As with Fasteners, I'll be happy to re-license to LGPL2.1. Again, one of the main components of the original WB was @ulrich1a 's sheetmetal unfolder which is LGPL2 so we are good with that.

Roy-043, 2023-12-07:
I have no objections to the LGPL2.1 license.

hpbmo, 2023-12-07:
I have also no objections to the LGPL2.1 license.

wavexx, 2023-12-07:
Thanks for pinging, I wouldn't have noticed otherwise. No objection to the license change!

fra589, 2023-12-07:
Hi all,  
I just made an action to correct a bug linked to the evolution of Python and FreeCAD, I do not feel entitled to decide on the licensing level...
Do as your wish :-)
Thanks for your work!

hasecilu, 2023-12-07:
Thanks for ping, it's nice to know that this WB could join upstream FreeCAD.
I agree with the license change, I'm also in for SheetMetal WB.

tobiasfalk, 2023-12-07:
In my opinion, my share of contribution is rather small and do not think that I would have a saying in this.
But I do not have a problem with this.

@prokoudine will there be an other Issue/Thread where one could give Suggestion of features that would be nice if they where implemented in this move to an Integrated workbench? This is since the most intelligent think to do, while moving this to FreeCAD, is to rewrite the core of this workbench in C++/Qt.

prokoudine, 2023-12-07:
@tobiasfalk I could be talking out of my arse, but I vaguely recall that @sliptonic and @shaise were in agreement that the unfolder shoud definitely be part of the core and thus be rewritten in C++. Once/if the relicensing of both WBs has green lights, there definitely should be review of both architecture, workflow, and UX/UI to come up with a sensible plan.

hasecilu, 2023-12-07:
(Copy the main post)
```md
- [X] @shaise
- [X] @alexneufeld
- [X] @riqueenz
- [ ] @FreeCAD-Tools
- [ ] @luzpaz
- [X] @hasecilu
- [ ] @berberic2
- [X] @Roy-043
- [ ] @EA32
- [ ] @elysium31
- [X] @kaktusus
- [X] @wavexx
- [X] @tobiasfalk
- [ ] @easyw
- [X] @fra589
- [ ] @wandrewkeech
- [ ] @algtgv
- [X] @hpbmo
- [ ] @chennes
- [ ] @ApexArray
- [ ] @JohanAR
- [ ] @elf128
- [ ] @chrido
- [ ] @spike77453
- [ ] @yorikvanhavre
- [ ] @f3nix
```

shaise, 2023-12-07:
@hasecilu, Thanks! great idea!

spike77453, 2023-12-07:
Works for me! :+1:

ApexArray, 2023-12-07:
No objections here 👍

algtgv, 2023-12-07:
No objection to the license change!

chennes, 2023-12-07:
No objection to the license change (and I think distributing this with FreeCAD would be a good idea :) )

chrido, 2023-12-07:
No objections to the license change :+1:

JohanAR, 2023-12-08:
No objections, I'm barely a contributor to this project anyway :)

JMG1, 2023-12-08:
No objections.

berberic2, 2023-12-10:
That is O.K. for me.

yorikvanhavre, 2023-12-11:
Sorry to be late :sweat_smile: I agree!

easyw, 2023-12-15:
go for it!
sorry being late, but I don't watch very often my GH notifications

berberic2, 2023-12-23:
FYIO: What `git blame` thinks about the missing user.
This is not a comprehensive analysis!

```
EA32 419 lines in 18 files
    GrammaticalTools.py:25
    InitGui.py:1
    FsData/gost1144head.csv:42
    FsData/gost1144length.csv:2
    FsData/gost1144range.csv:2
    FsFunctions/FSmakeWoodScrew.py:104
    Icons/ASMEB18.2.1.6.svg:14
    Icons/ASMEB18.21.1.12B.svg:11
    Icons/ASMEB18.3.5C.svg:12
    Icons/DIN508.svg:26
    Icons/GN507.svg:49
    Icons/GOST1144-3.svg:33
    Icons/GOST1144-4.svg:33
    Icons/ISO4014.svg:14
    Icons/ISO4028.svg:12
    Icons/ISO7089.svg:11
    Icons/ThreadedRod.svg:14
    Icons/ThreadedRodInch.svg:14

FreeCAD-Tools 1922 lines in 34 files
    TRANSLATIONS.md:73
    GrammaticalTools.py:41
    FastenerBase.py:29
    FastenersCmd.py:35
    README.md:4
    ScrewMaker.py:2
    translations/fasteners_pt-br.ts:182
    translations/fasteners_pt-pt.ts:182
    translations/fasteners_ru.qm:39
    translations/fasteners_ru.ts:885
    translations/update.sh:43
    translations/fasteners_es-ar.ts:137
    translations/fasteners_es-es.ts:137
    FsData/DiaList.csv:2
    FsData/din571length.csv:19
    FsData/gost1144length.csv:29
    FsData/gost1144range.csv:22
    FsData/gost11860def.csv:15
    FsFunctions/FSmakeCupNut.py:1
    FsFunctions/FSmakeWoodScrew.py:1
    Icons/ASMEB18.3.1A.svg:17
    Icons/ASMEB18.3.5A.svg:11
    Icons/ASMEB18.3.5B.svg:11
    Icons/ASMEB18.5.2.svg:13
    Icons/DIN96.svg:16
    Icons/GOST1144-1.svg:19
    Icons/GOST1144-2.svg:19
    Icons/GOST11860-1.svg:12
    Icons/ISO4026.svg:11
    Icons/ISO4027.svg:11
    Icons/ISO4762.svg:16
    Icons/PCBSpacer.svg:17

elysium31 0 lines in 0 files 0

wandrewkeech 0 lines in 0 files

elf128 0 lines in 0 files
```

elf128, 2023-12-23:
> waiting for response of:
> @riqueenz[https://github.com/riqueenz], @FreeCAD-Tools[https://github.com/FreeCAD-Tools], @luzpaz[https://github.com/luzpaz], @hasecilu[https://github.com/hasecilu], @berberic2[https://github.com/berberic2], @Roy-043[https://github.com/Roy-043], @EA32[https://github.com/EA32], @elysium31[https://github.com/elysium31], @kaktusus[https://github.com/kaktusus], @wavexx[https://github.com/wavexx], @tobiasfalk[https://github.com/tobiasfalk], @easyw[https://github.com/easyw], @fra589[https://github.com/fra589], @wandrewkeech[https://github.com/wandrewkeech], @algtgv[https://github.com/algtgv], @hpbmo[https://github.com/hpbmo], @chennes[https://github.com/chennes], @ApexArray[https://github.com/ApexArray], @JohanAR[https://github.com/JohanAR], @elf128[https://github.com/elf128], @chrido[https://github.com/chrido], @spike77453[https://github.com/spike77453], @yorikvanhavre[https://github.com/yorikvanhavre], @f3nix[https://github.com/f3nix]

Hey, sorry for late replly. I'm ok with the change.

shaise, 2023-12-23:
@berberic2 ,
I do not know how blame works but @elysium31 had several PRs. @wandrewkeech and @elf128 had one.
Perhaps their work was overwritten? 
Even so, they still count.
waiting for @FreeCAD-Tools  and @ea32 as well

berberic2, 2023-12-23:
As I said: This is not a comprehensive analysis!
blame gives for every line the last user that commited this line, so changing the indentation or whitespaces is sufficient.

elf128, 2023-12-23:
@elf128 is me. And, like I sad, I'm ok, with the change.
Once again, sorry, for late respose.

shaise, 2023-12-23:
@elf128 , thanks! noted!
@berberic2 , as I said perhaps their lines are no longer in the code, but they did contribute.

sliptonic, 2023-12-23:
If their changes are still there, we would proceed by removing them. So If their code changes are no longer present than we can proceed without their reply.

kaktusus, 2023-12-23:
@FreeCAD-Tools is a man from Ukraine. We all know the current situation in that country. Give him some time. :disappointed:

hasecilu, 2023-12-23:
```shell
git shortlog --author=FreeCAD-Tools --author=EA32 --author=elysium --author=keech
```

<details>
    <summary>Click to view commits</summary>
Andy (2):
      Add IUT[A/B/C] heatset inserts for 3D print modeling
      
Auto-sizing for IUTA/B/C heat sets

EA32 (9):
      GOST1144-3-4 fastener integration
      SVG icons optimization
      GOST 1144 head shape fix.
      Create head by B-Spline
      SVG icons optimization
      GOST 1144 (Wood screws) code refactoring
      Add text converter to singular form
      SVG icons optimization
      SVG icons optimization

FreeCAD-Tools (49):
      Full translation to Russian.
      Update Russian translation.
      Update Russian translation
      Added translate to treeview & fix LH flag for nuts
      Remove obsolete translations
      Minor Russian translation fixes.
      Integration of GOST11860 Type 1 Cap Nut
      DIN571 formating mistakes
      Add missing DIN7998 screw threads
      GOST1144 type 1 & type 2 integration
      Fix
      Fix2
      Update FastenersCmd.py
      Optimization of the command table.
      Update translation via Linux bash script
      Add files via upload
      Add files via upload
      Delete fasteners_uifiles_es-ar.qm
      Delete fasteners_uifiles_es-ar.ts
      Delete fasteners_uifiles_es-es.qm
      Delete fasteners_uifiles_es-es.ts
      Delete fasteners_uifiles_pt-br.qm
      Delete fasteners_uifiles_pt-br.ts
      Delete fasteners_uifiles_pt-pt.qm
      Delete fasteners_uifiles_pt-pt.ts
      Delete fasteners_uifiles_ru.qm
      Delete fasteners_uifiles_ru.ts
      Replace es_mx to es-mx and update script comments
      Delete fasteners_es_mx.qm
      Delete fasteners_es_mx.ts
      Minor fix update.sh
      Merge branch 'shaise:master' into master
      Optimize GOST 1144-1-2 and DIN96 SVG icons
      Fix 48x48px size
      Size fix
      PCBSpacer.svg optimization
      SVG icons optimization
      Update FastenerBase.py
      FastenerBase.py separate commands in code
      Update translation to Russian
      mx-es removed from list
      Change Scale B to Offset B
      Remove unused strings from ts files
      Remove unused strings from ts files
      Init Grammatical Tools
      SVG icons optimization
      Fix toolbars names
      Create HOW_TO_ADD_NEW_LANGUAGE.md
      Updating the article about translation

Maksym Veremchuk (10):
      Fix statements
      Add some more
      More
      Fix indentations
      Fix imports
      Fixes
      Works
      Merge branch 'master' into fix_indent
      More
      Refactor csv2dict
</details>

kaktusus, 2023-12-23:
> @FreeCAD-Tools is a man from Ukraine. We all know the current situation in that country. Give him some time. 😞

Or am I in the wrong?

wandrewkeech, 2024-01-15:
Hi! Sorry been afk on github for quite some time. 
This is great and fully support the license cha!nge

shaise, 2024-01-15:
great!
Waiting for @FreeCAD-Tools , @EA32 and @elysium31

elysium31, 2024-01-17:
No objections

leoheck, 2024-03-06:
Pinging @FreeCAD-Tools @EA32 

Could you guys comment here if this licence change is ok for you?

sliptonic, 2024-03-21:
I think we should post here the intention to change the license unless someone objects.  Leave it for 30 days or so and then proceed.  If either of the outstanding contributors complains, we can remove their contributions at a later date.

leoheck, 2024-03-21:
Why not just make a PR changing it and then if no one complains in by the end of the week. It gets merged, and if the complaints come after this, then their work can be removed as you said. There are just 2 contributors that are missing to respond. In other words, way to wait for a month? This was posted on December 2023.

sliptonic, 2024-03-21:
This isn't a huge emergency.  Changing the license won't immediately change anything.  Before we can consider including the WB in Ondsel or in upstream FreeCAD there's more work to be done and nobody is standing up to do it yet.  We have sheetmetal underway and many other important changes to work on.

tobiasfalk, 2024-03-23:
@shaise how much amd what did the two missing contribut.

shaise, 2024-03-23:
[image]
[image]
Quite a lot...

leoheck, 2024-03-23:
@shaise, Have you checked if they are active on GitHub still? _They may have died or been jailed..._

shaise, 2024-03-23:
@leoheck 
Already checked. They are not active. Really hope not for your proposed reasons...

tobiasfalk, 2024-03-24:
Two things.
First, I am no expert in Licensing but is this kind of decision somthing that requires approval fromm all or is this something a majority vote can decide?

Second, they both contributed quite a lot, is that something that is a major part of the WB or can it be easily redone, if a approval of all is required.

leoheck, 2024-03-25:
I asked ChatGPT to highlight the main differences between those two licences. I am not sure if this is correct tho. 

> Sure, I can explain the main differences between LGPL 2+ (Lesser General Public License version 2 or later) and GPL 2 (General Public License version 2).

1. **Scope of Application:**
   - **LGPL 2+:** Primarily designed for software libraries, allowing them to be used in proprietary software without requiring the proprietary software to be open source.
   - **GPL 2:** Designed for any kind of software, requiring any derivative work to be distributed under the same license.
2. **Derivative Works:**
   - **LGPL 2+:** Allows linking with proprietary software without forcing the proprietary software to become open source.
   - **GPL 2:** Requires any derivative work, including software linked with GPL-licensed code, to be distributed under the GPL license.
3. **Copyleft Requirements:**
   - **LGPL 2+:** Contains weaker copyleft provisions compared to GPL, allowing more flexibility in the licensing of derivative works.
   - **GPL 2:** Imposes strong copyleft provisions, ensuring that all derivative works are distributed under the same open source license.
  
> In simple terms, LGPL 2+ is like a "friendly" version of GPL 2 for software libraries. It allows those libraries to be used in both open source and proprietary software without imposing strict open source requirements on the proprietary software. On the other hand, GPL 2 applies to all types of software and requires any derivative work to be distributed under the GPL, promoting the open source nature of software development.

tobiasfalk, 2024-06-06:
Ping

kadet1090, 2024-08-27:
1. Commits from @FreeCAD-Tools are mostly related to SVG optimizations and translations hence the big numbers - I'm not sure if this is even copyrightable so maybe persmission is not needed. https://github.com/shaise/FreeCAD_FastenersWB/commit/1b0b5daa3e6a880a23ba89d0c58b90eb57ed60e3 this seems to be the only copyrightable change
2. Changes from @EA32 are mostly related to SVG and GOST1144 wood screw type - so we'll have to possibly remove and/or reimplement this part only

shaise, 2024-08-27:
The grand fault is actually mine. Back in the days, as a new github user I just next-clicked everything and end up with this faulty license. I believe all other developers did not care either when they contributed.
Since only 2 votes are missing and 8 months now with no response, I think we just need to change the license and leave here a message that if someone objects, let us know.

sliptonic, 2024-08-27:
I concur.  Let's move ahead.

tobiasfalk, 2026-11-12:
Hi, I am writing not only to maybe get this moving but also to represent an "alternative" idea if this cannot move forward.

Why not have this WB as its own repo (and keep its current license) and just have it installed by default? Or would that violate the license?

shaise, 2026-11-12:
I totally forgot about this issue since ondsel was canceled. I'm going forward with the license change. All but two agreed and the those 2 are inactive for 3 years now.

tobiasfalk, 2026-11-12:
> I totally forgot about this issue since ondsel was canceled. I'm going forward with the license change. All but two agreed and the those 2 are inactive for 3 years now.

Just to be certain, @FreeCAD-Tools and @EA32 if you see this, maybe say something.

And @shaise mabe set a deadline, so give them a week or so.

tobiasfalk, 2026-11-12:
And you should also check who else contributed since then, and if they have a problem with the change.
Since as you said it has been 3 Years, time flies.

shaise, 2026-11-12:
I'm just going to change the license, and then see if anyone objects, otherwise three more years will pass...

shaise, 2026-11-12:
License changed. Closing.
If there is any objection, please add a comment or reopen.