shaise/FreeCAD_FastenersWB #309: Aliases for quivalent norms

hasecilu, 2023-12-10:
Surfing on fasteners.eu I found:

**ISO 299 - Nuts for T-slots
Current norm: DIN 508
Equivalent norms: DIN 508; CSN 021529; PN 61275;** 

Could be useful if someone searches for ISO 299 without knowing that DIN 508 is equivalent, currently the DIN 508 norm is implemented so it would be nice to be redirected or use an alias to prevent duplication of data on the databases.

shaise, 2023-12-15:
Good Idea.
I have added support for that. see FSAliases.py

shai

hasecilu, 2023-12-16:
Thanks for the implementation, I checked it and seems to work mostly fine.
- I fixed a typo in the code.
- Added ISO299 to screwTables, without this name on tree is None and the BOM doesn't work

shaise, 2023-12-16:
@hasecilu 
>     * Added ISO299 to screwTables, without this name on tree is None and the BOM doesn't work
please see my comment on pr #312

hasecilu, 2023-12-16:
I tested it and it seems that the fasteners created with the alias ISO299 gets the name `None` with `name = FSRemoveDigits(obj.Name)` because in the BOM only the fasteners that were added using an explicit fastener are in the BOM.

- You can add a DIN508 and then change it to ISO299 and it will appear in the BOM
- If you add a ISO299 and then change it to DIN508  it will not appear in the BOM

shaise, 2023-12-16:
@hasecilu 
Ah it seems I misplaced my screwTables update function. please try again.

shai

hasecilu, 2023-12-16:
Cool, now it's working fine, thanks!

shaise, 2023-12-16:
Thanks! closing.

closed 2023-12-16 by shaise