shaise/FreeCAD_FastenersWB #315: [Problem] Countersunk holes feature is no longer needed

alexneufeld, 2023-12-25:
The [PartDesign_Hole](https://wiki.freecad.org/PartDesign_Hole) tool now supports everything that this tools can do and more. In the interest of not re implementing functionality present in the base FreeCAD codebase, I propose we deprecate this feature immediately, and remove it entirely at the next major FC release.

Thoughts?

[image]

shaise, 2023-12-25:
does the new tool knows the dimensions needed for the countersunk hole per screw size and head type?

hasecilu, 2023-12-25:
As user I have been always using the countersunk feature from PartDesign.

It seems it has a few definitions from DIN/ISO norms.

[image]

alexneufeld, 2023-12-25:
yes, it actually supports everything that the FastenersWB tool does: 
[image]
``` python
>>> import ScrewMaker
>>> x = ScrewMaker.FSScrewMaker()
>>> x.GetAllCountersunkTypes()
['ISO10642', 'ISO14582', 'ISO14584', 'ISO2009', 'ISO2010', 'ISO7046', 'ISO7047']
>>> 
```

ISO2010 and ISO7047 should not have different countersink geometry than ISO2009 and ISO7046 respectively.

tobiasfalk, 2023-12-26:
I don't know why it should be removed, maybe a text that it is not recommended to use it, but just removing it will most likely lead to more problems.

alexneufeld, 2023-12-26:
It should be removed because:
- It's less robust and slower than the PartDesign_Hole tool
- The GUI for it isn't up to any kind of UX standard. The user is just shown a checklist of all possible edges, with no filtering or context. 
- There should only be **one** obvious way to accomplish any particular task. As a widely used community addon, we have a responsibility to follow this principle.

This change will break things. **That's the point**. We'll first add a warning to the report view when restoring a countersunk hole python object. Users who still actively work with this feature **will** be given the knowledge of this change and how to adapt to it.

shaise, 2023-12-26:
The tool in part design seems much more comprehensive, so indeed I agree it should be deprecated in fasteners. The fasteners one is also not compatible with bodies.
However, I suggest that it should only be removed from the UI such that new files will not be able to use it. This way old files can be loaded non broken.

PR #346 (2024-02-25): Deprecate CountersunkHoles object
Closes #315

Print a deprecation warning when restoring documents that use this object.

The GUI command to add this feature to new designs is removed, but existing objects will continue to function properly
commit 5c55de9d (2024-02-25): Deprecate CountersunkHoles object

Closes #315

Print a deprecation warning when restoring documents that use this
object.

The GUI command to add this feature to new designs is removed, but
existing objects will continue to function properly
closed 2024-02-25 by shaise