shaise/FreeCAD_FastenersWB #318: List of pins

hasecilu, 2023-12-27:
I made a few pins and saw other ISO norms, so I made a list with them.

Non-exhaustive list of implemented types of pins.

Video-reference: https://youtu.be/W-0ZV9zXpc0         

### Clevis pins                                           
- [X] Single hole clevis pins, w/wo head - **ISO2340**, **ISO2341-[A/B]**
- [ ] Adjustable clevis pins                            
- [ ] Quick release detent clevis pins                  
- [ ] Clevis pins with grooves                             
### Dowel pins                                            
- [X] Plain dowel pins **ISO2338**
- [ ] Dowel pins - ISO8734                              
- [ ] Parallel Pins with Internal Thread - ISO87[33/35] 
- [ ] Tapped dowel pins                                 
- [ ] Stepped dowel pins                                                          
### Taper pins                                            
- [X] Plain taper pins - **ISO2339**
- [ ] Taper pins with Internal Thread - ISO8736         
- [ ] Taper pins with external thread - ISO8737         
- [ ] Threaded taper pins                               
- [ ] Threaded taper pins  
### Roll pins                                             
- [ ] Slotted Roll tension pins (C shape) - ISO8752     
- [ ] Toothed slotted spring pins (skew proof)          
- [ ] Spiral dowel pins, heavy duty - ISO87[48/50]      
### Grooved pins                                          
- [ ] Full length parallel grooved pins                 
- [ ] Full length taper grooved pins - ISO8744          
- [ ] Half length taper grooved pins - ISO8745          
- [ ] Half reverse taper grooved pins                   
- [ ] Third length centre grooved pins - ISO8742        
- [ ] Half length centre grooved pins - ISO8743         
- [ ] Round Head Grooved Pins - ISO8746                 
- [ ] Countersunk Head Grooved Pins - ISO8747           
- [ ] Headed grooved pins

alexneufeld, 2023-12-27:
Note that I have this branch almost ready to merge. [https://github.com/alexneufeld/FreeCAD_FastenersWB/tree/metric_pins_merge](https://github.com/alexneufeld/FreeCAD_FastenersWB/tree/metric_pins_merge)

It has every ISO standard pin that I could find

hasecilu, 2023-12-27:
Oops, https://github.com/hasecilu/FreeCAD_FastenersWB/tree/pins

Do we have and official thread on the forum for WIP?

alexneufeld, 2023-12-27:
No forum thread. I just need to fix up the diameter matching functionality and we'll merge it ASAP. shouldn't take long

hasecilu, 2023-12-27:
I tested you branch, looks very good!
- I would just change in the descriptions the first letter to be uppercase (to be consistent with all other descriptions)
- In the ISO2341 there is **A** (no hole) and **B** (with hole) types, should be both offered?

PR #338 (2024-02-12): Add Metric pins
closes #318

Auto-Diameter matching is TBD for pins. That bit of code needs cleaning up for some of the recently added fastener types. I'll get to it, just not today.
closed 2024-02-12 by shaise