shaise/FreeCAD_FastenersWB #319: [Feature request] Custom Designation string for each norm

tobiasfalk, 2024-01-02:
An ISO norm typically states how the specified part should be designated, it would be nice if one could specify a template string in the function, so that the FreeCAD label is automatically in the norm designation form.
I was sadly not able to find one norm where a naming convention is specified, but the ISO and DIN norms all have a similar, if not identical, style. 

- ISO 8676: M12x1,5 l = 80mm "Festigkeitsklasse" 8.8 is a **Sechskantschraube ISO 8676 - M12 x 1,5 x 80 - 8.8**
- ISO 4032: Typ 1 d = M12  "Festigkeitsklasse" 8 is a **Sechskantmutter ISO 4032 - M12 - 8**
- DIN 961: M8x1 l = 50mm material assigned property class 8.8 is a **Hexagon head bolt DIN 961 - M8 x 1 x 50 - 8.8**
- DIN 961: M8x1 l = 50mm material assigned property class 8.8 product grade A is a **Hexagon head bolt DIN 961 - M8 x 1 x 50 - 8.8 - A**

Yes, the Material and a Production grade are mentioned in there. 
The material class could be created as a Parameter.
The grade is something that would need the "duplication" of some norms or another more complicated implementation, but this is another topic. (simply put many have multiple grades where the table would need to be done twice for some sizes, how one selected between two is not this issue)

tobiasfalk, 2024-01-02:
for ASME, ASME B18.12.1 - 2006 A Guide for the Development
of
ASME B18 Standards
for
Mechanical Fasteners (2.6 Designation on page 8) seams to define a standard method of designation, but I think a pure global designation style may not the best solution.

hasecilu, 2024-01-03:
That would be nice for the BOM having "Hexagon head bolt DIN 961 - M8 x 1 x 50 - 8.8" but in the Tree View maybe something shorter like "DIN 961 - M8x1x50".
Respect to materials: a lot of work has been done on the Materials WB so maybe when it's done could be implemented here, personally I haven't tested it already.

tobiasfalk, 2024-01-03:
> Respect to materials: a lot of work has been done on the Materials WB so maybe when it's done could be implemented here, personally I haven't tested it already.

I do not think so, because the standards typically define in what property class and material the "thing" is available.
ISO 8676 states that with Steel there is 5.6; 8.8 and 10.9 for d <= 39mm and with Stainless steel there is A2-70; A4-70 for d <=24mm, A2-50; A4-50 for 24mm < d <= 39mm. (also some more, but this is just for illustration)
The same is with the other DIN and ISO standards.
So the possibility to add a data table would be more useful for the property class, but a Simple string property would be a good start, maybe the ability to add a standard depending default value.

tobiasfalk, 2024-01-03:
Look here: https://www.bigboltnut.com/product/iso-4014.html at table **Specifications and reference International Standards**