shaise/FreeCAD_FastenersWB #320: DIN 1624 is wrongly named

tobiasfalk, 2024-01-03:
[DIN 1624](https://www.beuth.de/en/standard/din-1624/1349780) is something completely different(Steel flat products; cold reduced mild unalloyed steel strip in widths not exceeding 650 mm; technical delivery conditions).
It is also not [ISO 1624](https://www.beuth.de/en/standard/iso-1624/50214523).
I was not able to find a corresponding standard for it, but I think it should be renamed.
I also saw that if one searches DIN 1624 results for this appear, but not many.

tobiasfalk, 2024-01-03:
I just realized that it is in FSDatata and [FastenersCmd.py](https://github.com/shaise/FreeCAD_FastenersWB/blob/a0810d5311dac34413cd2c60158c23e4c3cf5bbf/FastenersCmd.py#L176) but not available in FreeCAD

alexneufeld, 2024-01-21:
You are likely correct that these T-nuts aren't actually specified by the DIN 1624 standard.
Several manufacturers do list them as such, however (which is why I initially labeled them this way):
- https://fullerfasteners.com/products/din-1624-t-nut-bare-with-4-prongs/
- https://www.singhaniainternational.com/products-details/T-Nut_-Four_Prong_Tee-Nut-_DIN_1624/105
- https://www.xinchi-industry.com/din-1624/

Most likely the steel stock they are made from conforms to the *actual* DIN 1624 standard?? This seems to be an annoying case where common industry practice is nonsense, and there's no actual standard to follow. Very annoying for a simple and reasonably commonly used fastener...

tobiasfalk, 2024-01-22:
We already have non standard parts that have a descriptive name, so doing the same here sounds reasonable.
Somthing on the line of "4 prong wood thread insert".
While doing that I would suggest moving it in to the Insert category, because if I am right these are used for wood as inserts.

alexneufeld, 2024-01-22:
sounds good to me :+1:

PR #329 (2024-02-05): DIN 1624 renaming to 4 Prong Wood Thread Insert (4PWTI)
This is to resolve #320.
The renaming is done but beckwards compabilaty is not given. I have tried multiple things like adding 
`if obj.type == "DIN1624":`
`            # backward compatibility - remove after Issue #320`
`            obj.type = type = "4PWTI"`
to FastenersCmd.py, but I get: 
`<class 'ValueError'>: '4PWTI' is not part of the enumeration in DIN1624#Nut.type`
And it does not update it.
closed 2024-02-05 by shaise