shaise/FreeCAD_FastenersWB #321: List of DIN Fasteners

tobiasfalk, 2024-01-03:
Still need sorting but here is a list of DIN standards and there coresponding ISO standard with chages.
The list is base on [Fuller](https://fullerfasteners.com/tech/din-iso-en-crossover-chart/)

- [ ] DIN 1 Unhardened taper pins **; ISO 2339, Nominal length redefined (with round end)**
- [ ] DIN 7 Parallel pins of unhardened steel and austenitic stainless steel **; ISO 2338, Nominal length redefined (with round end) form A+C deleted**
- [X] DIN 84 Slotted cheese head screws – Product grade A **; ISO 1207, Modifications of dimensions < 2,5; dimension of slots changed; partly nominal lengths deleted; strength 8.8 deleted**
- [ ] DIN 85 Slotted pan head screws – Product grade A **; ISO 1580, Thread included; dimensions of head and slot changed; strength 8.8 deleted**
- [ ] DIN 94 Slotted set screws with long dog point **; ISO 1234, No modifications of dimensions**
- [X] DIN 96 Slotted round head wood screws

- [ ] DIN 125-1 Product grade A washers with a hardness up to 250 HV, designed for use with hexagon head bolts and nuts **; ISO 7089, ISO 7090, Without chamfer, up to nominal diameter 64**
- [ ] DIN 125-2 Product grade A washers with a hardness from 300 HV, designed for use with hexagon head bolts and nuts **; ISO 7089, ISO 7090, Without chamfer, up to nominal diameter 64**
- [ ] DIN 126 Product grade C washers, designed for use with hexagon head bolts and nuts **; ISO 7091, Nominal sizes changed**

- [X] DIN 315 Fasteners - Wing nuts - Rounded wings 

- [ ] DIN 417 Slotted set screws with long dog point **; ISO 7435, No modifications of dimensions**
- [ ] DIN 427 Slotted headless screws with shank **; ISO 2342**
- [ ] DIN 433-1 Product grade A washers with a hardness up to 250 HV, designed for use with cheese head screws **; ISO 7092, Pyramid diamond hardness 140 HV cancelled; rating changed**
- [ ] DIN 433-2 Product grade A washers with a hardness up to 300 HV, designed for use with cheese head screws **; ISO 7092, Nominal sizes changed**
- [ ] DIN 439-1 Hexagon thin nuts with M 1,6 to M 10 threads **; ISO 4036, Correlation of dimensions with DIN 439-1, except for the width across flat of the nut M10**
- [ ] DIN 439-2 Hexagon thin nuts with M 1,6 to M 52 and M 8×1 to M 52×3 threads **; ISO 4035, ISO 8675, In tables 1 and 2 abbreviations changed and some specifications (thread dimensions) changed, Standard divided by regular type screw threads and fine threads; width across flats according to ISO 272 changed; nominal size range changed; fine thread see ISO 8675; regular type screw thread see ISO 4035**
- [ ] DIN 440 Plain washers – Extra large series, product grade C **; ISO 7094, Nominal sizes changed, nominal size 24 size “a” redefined to the maximum, square-end hole Form R omitted; (standard DIN 440 still valid)**
- [X] DIN 471 Retaining rings for shafts - Normal type and heavy type
- [X] DIN 472 Retaining rings for bores - Normal type and heavy type
- [X] DIN 478 Square head bolts with collar

- [X] DIN 508-1 Roofing and cladding products from metal sheet - Specification for self-supporting products of steel, aluminium or stainless steel sheet - Part 1: Steel **; t-Nut exists in FSData but not selectible in FreeCAD**
- [X] DIN 508-2 Roofing and cladding products from metal sheet - Specification for self-supporting products of steel, aluminium or stainless steel sheet - Part 2: Aluminium **; t-Nut exists in FSData but not selectible in FreeCAD**
- [X] DIN 508-3 Roofing and cladding products from metal sheet - Specification for self-supporting products of steel, aluminium or stainless steel sheet - Part 3: Stainless steel **; t-Nut exists in FSData but not selectible in FreeCAD**
- [ ] DIN 551 Slotted set screws with flat point **; ISO 4766, No modifications of dimensions**
- [ ] DIN 553 Slotted set screws with cone point **; ISO 7434, No modifications of dimensions**
- [ ] DIN 555 Hexagon nuts; Metric Thread, Type g **; ISO 4034, Width across flats according to DIN ISO 272 included; nominal diameter reduced**
- [ ] DIN 558 Hexagon screws, Thread Approximately to Head, Metric Thread, Type g **; ISO 4018, Nominal size extended; width across flats according to DIN ISO 272 included**
- [X] DIN 571 Hexagon head wood screws

- [ ] DIN 601 Hexagon Bolts without Hexagon Nut – with Hexagon Nut, Type g **; ISO 4016, Nominal size extended; width across flats according to DIN ISO 272 included**
- [X] DIN 603 Mushroom head square neck bolts **; ISO 8677**

- [ ] DIN 912 Hexagon socket head cap screws **; ISO 4762, Nominal size range changed; fine thread deleted (ISO 21269)**
- [ ] DIN 913 Hexagon socket set screws with flat point **; ISO 4026, Nominal sizes M1.4, M1.8, M14, M18 and M22 omitted**
- [ ] DIN 914 Hexagon socket set screws with cone point **; ISO 4027, Nominal sizes M1.4, M1.8, M14, M18 and M22 omitted**
- [ ] DIN 915 Hexagon socket set screws with dog point **; ISO 4028, Nominal sizes M1.4, M1.8, M14, M18 and M22 omitted**
- [ ] DIN 916 Hexagon socket set screws with cup point **; ISO 4029**
- [X] DIN 917 Hexagon cap nuts, low type
- [X] DIN 928 Square weld nuts
- [X] DIN 929 Hexagon weld nuts
- [ ] DIN 931-1 Hexagon head bolts with shank; M 1,6 to M 39 screw threads, Product grades A and B **; ISO 4014, Nominal size extended; width across flats according to DIN ISO 272 included**
- [X] DIN 933 Hexagon head screws – Product grades A and B **; ISO 4017, Nominal size extended; width across flats according to DIN ISO 272 included**
- [X] DIN 934 Hexagon nuts; Metric screw threads; Product grades A and B **; ISO 4032, ISO 8673, Nominal diameter changed, height of the nut changed partially; nuts with fine thread specified in DIN EN ISO 8673; nuts with regular type screw thread specified in DIN EN ISO 4032; width across flats according to DIN ISO 272 included**
- [ ] DIN 935-1 Hexagon slotted and castle nuts - Part 1: Metric coarse and fine pitch thread, product grades A and B **; only coarse pitch is implemented, the fine pitch is missing**
- [ ] DIN 935-3 Hexagon slotted nuts - Part 3: Metric coarse pitch thread, product grade C
- [ ] DIN 960 Hexagon bolts, Metric fine thread, Types m and mg **; ISO 8765, Nominal size extended; width across flats according to DIN ISO 272 included**
- [X] DIN 961 M 8×1 to M 100×4 hexagon head bolts with fine pitch thread **; ISO 8676, Nominal size extended; width across flats according to DIN ISO 272 included**
- [ ] DIN 963 Slotted countersunk (flat) head screws (Countersunk heads according to ISO) **; ISO 2009, Nominal size changed; head dimension changed; thread length changed; strength 8.8 cancelled**
- [ ] DIN 964 Slotted raised countersunk (oval) head screws (Countersunk heads according to ISO) **; ISO 2010, Nominal size changed; head dimension changed; thread length changed; strength 8.8 cancelled**
- [ ] DIN 965 Cross recessed countersunk (flat) head screws (Countersunk heads according to ISO) **; ISO 7046-1, Nominal size changed; head dimension changed; thread length changed; strength 8.8 included (see Part 2); penetration depth of cross slots changed**
- [ ] DIN 966 Cross recessed countersunk (oval) head screws (Countersunk heads according to ISO) **; ISO 7047, Nominal size changed; head dimension changed; thread length changed; strength 8.8 omitted; penetration depth of cross slots changed**
- [X] DIN 967 Cross recessed pan head screws with collar - Product grade A
- [ ] DIN 980 All-metal prevailing torque type hexagon nuts **; ISO 7042, Height of nut changed; width across flat according to ISO 272 included; and ISO 10513, Height of nut changed; width across flat according to ISO 272 included**
- [ ] DIN 982 Prevailing torque type hexagon nuts with non-metallic insert **; ISO 7040, ISO 10512, Height of nut changed; width across flat according to ISO 272 included**
- [ ] DIN 985 Prevailing torque type hexagon thin nuts with non-metallic insert **; ISO 10511, Height of nut changed; width across flat according to ISO 272 included**

- [ ] DIN 1433 Clevis pins without head **; ISO 2340, Some other nominal lengths defined**
- [ ] DIN 1434 Clevis pins with head **; ISO 2341, Only one head dimension. Some other nominal lengths defined**
- [ ] DIN 1435 Clevis pins with head **; ISO 2341, Only one head dimension. Some other nominal lengths defined**
- [ ] DIN 1436 Clevis pins with head **; ISO 2341, Only one head dimension. Some other nominal lengths defined**
- [ ] DIN 1440 Plain washers for clevis pins; product grade A **; ISO 8738, Nominal sizes changed**
[... cut ...]
- [X] DIN 7984 Hexagon socket head cap screws with low head with reduced loadability
- [ ] DIN 7985 Recessed raised cheese (fillister) head screws **; ISO 7045, Nominal size changed; head dimension changed; thread length changed; penetration depth of cross slots changed**
- [ ] DIN 7991 Hexagon socket countersunk head screws **; ISO 10462**
- [X] DIN 7996 Cross recessed round head wood screws

- [ ] DIN 9021 Washers; outside diameter ˜ 3 x bolt diameter **; ISO 7093-1, ISO 7093-2, Dimensions partially changed**

hasecilu, 2024-01-03:
Fasteners with no changes in dimensions could be implemented as alias and it's very easy to maintain and for those which have extra nominal sizes a new feature to extend another fastener type may be needed to avoid duplicating common sizes.
Isn't the DIN 508 T-Slot nut working for you?

tobiasfalk, 2024-01-03:
> Fasteners with no changes in dimensions could be implemented as alias and it's very easy to maintain and for those which have extra nominal sizes a new feature to extend another fastener type may be needed to avoid duplicating common sizes. 

Yes sounds good, but checking if it really is the same still needs to be done.
I did not confirm everything that is written as change

tobiasfalk, 2024-01-03:
>  Isn't the DIN 508 T-Slot nut working for you?

Yes I do not see it in FreeCAD it self, the same is with "DIN 1624" thats why I closed the issue direktly #320

hasecilu, 2024-01-03:
It's on its own group "TSlot", it won't appear in the list in other nuts
[image]

tobiasfalk, 2024-01-03:
Ok, had to remove and re-install it, and now it appears

hasecilu, 2024-03-06:
With #353 you can mark 1143, 1144, 1151, 1152, 1160, 464, 465, 653

ToniTen, 2024-09-04:
DIN 25201 wedge lock fasteners are not included in this list. I requested them on #400

tobiasfalk, 2024-09-06:
@ToniTen do you mean

- [] DIN 25201-1 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 1: Einteilung, Kategorien der Schraubenverbindungen
- [] DIN 25201-2 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 2: Konstruktion - Maschinenbauliche Anwendungen
- [] DIN 25201-3 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 3: Konstruktion - elektrische Anwendungen
- [] DIN 25201-4 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 4: Sichern von Schraubenverbindungen
- [] DIN 25201-5 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 5: Korrosionsschutz
- [] DIN 25201-6 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 6: Anschlussmaße
- [] DIN 25201-7 Konstruktionsrichtlinie für Schienenfahrzeuge und deren Komponenten - Schraubenverbindungen - Teil 7: Montage

I think that part 6 has the dimensions, but I am not sure size I do not have access to those

Rahix, 2025-05-17:
ISO 4762 / DIN 912 is already present :)

tobiasfalk, 2025-05-19:
> ISO 4762 / DIN 912 is already present :)

 Yes and no, DIN 912 has M1,4, M18, M22, M27, M33, M42, M48, M56, M64, M72, M80, M90 and M100

In #286 it is checked of but in my opinion for DIN 912 the missing sizes need to be added to a DIN 912 type.