shaise/FreeCAD_FastenersWB #323: Info-Display broken in AddonManger

berberic2, 2024-01-30:
In the current git-version the display of the information-page in **_AddonManager_** is broken.
Possibly because of the `‣ Expand this section for a synopsis on how to use this workbench` feature.
[image]

alexneufeld, 2024-02-25:
I can't reproduce this issue on a recent weekly build

[image]

```
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 (Git) AppImage
Build type: Release
Branch: main
Hash: ed10fe216d332debfd317d4639e284c50f5dab47
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * btl 0.9.9
  * FreeCAD-Flat-Icons
  * OpenDark 2023.12.1
  * TDAlignItems
  * Curves 0.6.22
  * fcweld 0.1.0
  * FreeCAD_SheetMetal 0.4.1
  * fasteners 0.5.11
```

berberic2, 2024-02-25:
Still broken here.
I will pull from git and rebuild to verify.

shaise, 2024-02-25:
confirmed on windows 10: works on version 0.21.2, not working on latest weekly.

Looks like an addon manager issue.

luzpaz, 2025-02-10:
This should be fixed by now. Closing.

closed 2025-02-10 by luzpaz