shaise/FreeCAD_FastenersWB #324: [feature] nails

Anthony-Gaudino, 2024-01-30:
I wonder if there are any plans for adding nails to the FastenersWB.

I was looking into using nails in an assembly, but I could not find it on FastenersWB.
Woodworking workbench also doesn't have nails.

I end up creating my own nail.

I think having nails on FastenersWB would be useful.

hasecilu, 2024-01-30:
Do you know about any norms about it, mainly we are interested on the dimensions to create the geometries.

hasecilu, 2024-01-30:
The most boring task is to recolect data, most of time the code to create the objects are trivial, are you interested on creating the csv files? I'm already busy with some thumb screws.

Anthony-Gaudino, 2024-01-31:
I can help creating the CSV files.

Should these files be placed directly into the `FsData` directory?

The CSV should follow this format right? (example for Steel Common Nails):
```
"F 1667 NLCMS"
"DN","S","Len","Dia","Head_dia"
"01","2d","1in","0.072in","0.172in"
```

Field reference:
- DN: dash number, used to reference size in ID, example: `F 1667 NLCMS 01`
- S: trade designation (reference in penny weight), used to reference size, should be added in menu for user to pick nail size
- Len: Shank length
- Dia: Shank diameter
- Head_dia: head diameter

Is there any pitfalls I should avoid?
If values in specification are available in both inch and mm, should I use mm instead of inch?

hasecilu, 2024-01-31:
Yes, on `FsData` with the name of the norm.
The most important dimension is the diameter so it should be the first column.

If for every diameter there is just a single length it could be all in the same file but if there are various lengths for each diameter we need 3 files: `def`, `length` and `range`.

The diameter should be in inches because is a imperial norm, then a dictionary is used to convert inch-diameter "keys" to "float" metric (mm) numbers, all other dimensions should be given in mm. Example or quarter inch:

```csv
"ASMEB18.2.1.6length"
"Nominal","Min","Max"
"1/4in",6.1595,6.5405
```

Anthony-Gaudino, 2024-01-31:
Is this acceptable?
```
"F 1667 NLCMS",,,,
"Dia",Len,Head_dia,DN,S
"0.072",1,0.172,1,2d
"0.08",1.25,0.203,2,3d
"0.099",1.5,0.25,3,4d
"0.099",1.75,0.25,4,5d
"0.113",2,0.266,5,6d
"0.113",2.25,0.266,6,7d
"0.131",2.5,0.281,7,8d
"0.131",2.75,0.281,8,9d
"0.148",3,0.312,9,10d
"0.148",3.25,0.312,10,12d
"0.162",3.5,0.344,11,16d
"0.192",4,0.406,12,20d
"0.207",4.5,0.438,13,30d
"0.226",5,0.469,14,40d
"0.244",5.5,0.5,15,50d
"0.262",6,0.531,16,60d
```

tobiasfalk, 2024-02-01:
> From what I could find, there's a metric standard. US follows a different standard.

There is:
- DIN 1143-1 Nails for use in automatic nailing machines (withdrawn)
- DIN 1144 Nails for light weight building slabs (withdrawn)
- DIN 1151 Round plain head nails (withdrawn)
- DIN 1152 Round nails with cold headed head (withdrawn)
- DIN 1158 Hook nails (withdrawn)
- DIN 1160 Wire nails with extra large head (withdrawn)
- DIN 1163 Molding pins (withdrawn)
- DIN 68163 Fasteners for assembling timber pallets; nails (withdrawn)

- DIN EN 10230-1 Steel wire nails - Part 1: Loose nails for general applications (ex DIN 1151, there may be changes)

hasecilu, 2024-02-01:
> Is this acceptable?
All dimensions in mm, (fake data)
```csv
"F-1667-NLCMSdef"
"DN","Dia","Len","Head_dia","S"
"01",0.023,12.23,5,6
```

From what I see in the data every dash number represents a single nail, which means a single diameter and a single length, **is this true?** In that case the dash number should be the first column.
Normally we use this method to get the diameter in mm because the diameter used to be M3, 1/4in, ST6.3, etc. Check `FsData/DiaList.csv` file
```python
    dia = self.getDia(fa.calc_diam, False)
```
In that case should be:
```python
    d, l, hd, s = FsData["F-1667-NLCMSdef"][fa.calc_diam] # fa.calc_diam is the dash number key
```

Anthony-Gaudino, 2024-02-02:
> All dimensions in mm, (fake data)

I didn't understand what you mean when you say **fake data**.
Do you mean I need to put the `in` unit in the cell like `2in`?
Or that the data I sent before has fake values? The values are from the standard.

You said before that the diameter should be in inches, and the other values in mm, so my previous CSV is wrong, as all values are in inches.
You said that it should be in inches because it's an imperial norm, but in the norm, it provides values in both inches and mm.

I saw `FsData/DiaList.csv` and it has mostly inch values in fractions, like `1/4`, but the ASTM standard provides the values in decimals, both for inches and millimeters.

I think that if internally the library will convert from inches to millimeters, then maybe the CSVs could have all values in mm.

--------

> From what I see in the data every dash number represents a single nail, which means a single diameter and a single length, **is this true?**

Yes. Some dash numbers may share same value for one or more dimensions, for example `F 1667 NLCMS-03` and `F 1667 NLCMS-04` have the same shank diameter of `2.51"`.

hasecilu, 2024-02-02:
> I didn't understand what you mean when you say **fake data**.

My bad I meant the data I put doesn't come from the docs, I invented them. :sweat_smile: 

> You said before that the diameter should be in inches, and the other values in mm, so my previous CSV is wrong, as all values are in inches.
> You said that it should be in inches because it's an imperial norm, but in the norm, it provides values in both inches and mm.

The data of the geometry is determined by the diameter and length, this 2 parameters are exposed to the user so they can choose a 1/4in x 3/8 in set screw instead of a 6.35 x 9.525 one, so that's why this values should be in inches. Since all other data is not showed to user all of them is in mm because that's the default, there are functions working to convert diameters and lengths.

The data from second column and above should be in mm.

> Yes. Some dash numbers may share same value for one or more dimensions, for example `F 1667 NLCMS-03` and `F 1667 NLCMS-04` have the same shank diameter of `2.51"`.

If I understand correctly it's a single norm that manage different geometries, I would recommend to create a single table for each geometry, data may be duplicated.

You can create a single file called "F-1667-NLCMSdef.csv" and inside put various data tables separated by a blank line (check iso7380def.csv as example).

```
"F-1667-NLCMS-01def"
"DN","Dia","Len","Head_dia","S"
"01",0.023,12.23,5,6

"F-1667-NLCMS-02def"
"DN","Dia","Len","Head_dia","S"
"01",0.023,12.23,5,6
```

and then use `d, l, hd, s = FsData["F-1667-NLCMS-01def"][fa.calc_diam]`

Anthony-Gaudino, 2024-02-02:
OK, so I can create a single file for a type of nail, for example:

**F-1667-NLCMdef.csv** for common nails, then inside I can add the specification data for `F 1667 NLCMA` (Aluminum Common Nails), `F 1667 NLCMC` (Copper Common Nails) and `F 1667 NLCMS` (Steel Common Nails).

And then another file **F-1667-NLBXdef.csv** for box nails, with `F 1667 NLBXA` and `F 1667 NLBXB` specification data.

-----

It's still not clear to me if I can add all data as mm.

For ASTM standard nails, users would select the size by trade designation (examples: 2d, 6d), so it's not the same as for threaded screws where users pick a dimension like `1/4"`.

hasecilu, 2024-02-03:
> **F-1667-NLCMdef.csv** for common nails, then inside I can add the specification data for `F 1667 NLCMA` (Aluminum Common Nails), `F 1667 NLCMC` (Copper Common Nails) and `F 1667 NLCMS` (Steel Common Nails).

If they have different geometries that would make sense

> It's still not clear to me if I can add all data as mm.

Yes, you should use mm.

> For ASTM standard nails, users would select the size by trade designation (examples: 2d, 6d), so it's not the same as for threaded screws where users pick a dimension like `1/4"`.

What 2d, 6d means? 2/6 times the diameter? is the diameter given?

For example in  TABLE 3 Type I, Style 1—Brads:
If I want a 2in nail, let say I choose Dash No. 26, `L` and `D` are given what is 6d?
Dash number is supposed what designer choose isn't? So that data should be first column and it's show to user.

Anthony-Gaudino, 2024-02-03:
OK, I think I got it mostly figured out.

Not all nail types have the trade designation (reference in penny weight), like d2, d6, etc. The penny is used to identify the length of the nail, and is mostly used only in the US. The use of penny weight, it seems, is diminishing and being replaced by just having the length.

So, with this, we could skip having the penny weight, and instead use the length directly. Using the length is the same as FastenersWB has been using for threaded screws, but instead of it being the width, it's the length for nails. And the values are similar, like `1/4in` length nail, so I can add the length as first column in inches like it's for screws.

### How users would add a nail

* Users would select in the menu the type of nail by clicking on the icon (common, box, sinker...).
* Then on the data panel, they could select the size (diameter parameter for threaded screws, but will be length for nails). _One problem here, some nail types like `common` have different subtypes (for common nails: Aluminum, Copper, Steel) so we could either:_
  1. Have in the data panel along with the length, another parameter to select which subtype it is.
  2. When selecting the length, have entries like `aluminum 1/4in` so it's both the subtype and the length as single parameter.
  3. Have totally separate menu entries for each subtype, so one icon for each subtype in the panel.
  4. Have a submenu in the main menu for each subtype, when clicking on the icon a submenu would appear where the user could pick a subtype.

I think these parameters would not be used for nails and thus should not be available if the fastener is a nail:
- left handed
- length
- length custom
- match outer
- thread

There should be other options for nails, for example:
- Tip style - Some nail types support different tip styles
- Head pattern - Some nails have patterns on the head, for example a checkered pattern
- T-head - Nails can have a T-head, where 2 sides of the head are cut.
- Clipped head - Where a part of the nail head is cut.

-------

Here's example data for Steel common nail:
```
"F 1667 NLCMS",,
"Len",D,H
"1in",1.83,4.37
"1 1/4in",2.03,5.16
"1 1/2in",2.51,6.35
"1 3/4in",2.51,6.35
"2in",2.87,6.76
"2 1/4in",2.87,6.76
"2 1/2in",3.33,7.14
"2 3/4in",3.33,7.14
"3in",3.76,7.92
"3 1/4in",3.76,7.92
"3 1/2in",4.11,8.74
"4in",4.88,10.31
"4 1/4in",5.26,11.13
"5in",5.74,11.91
"5 1/2in",6.2,12.7
"6in",6.65,13.49
```
`Len` is the length in inches, `D` is the diameter in mm and `H` is the head diameter in mm.

-----

Does this make sense?
How should the different subtypes be handled?

tobiasfalk, 2024-02-03:
But the length is not a number that is definetif, F1667-NLCMMS (Table 16) has one length multiple times but with different diameters.
I think having the Dash No. in front is the best way, also because that is how the standard differentiates the sizes. That's the job of the Dash No.

tobiasfalk, 2024-02-03:
With the DIN standards it is a combination of diameter x length (10x15 for a 1mm diameter and a 15mm long nail)

Anthony-Gaudino, 2024-02-03:
> But the length is not a number that is definetif, F1667-NLCMMS (Table 16) has one length multiple times but with different diameters. I think having the Dash No. in front is the best way, also because that is how the standard differentiates the sizes. That's the job of the Dash No.

Yes, you're right, it's not possible to identify it by just the length then, but having the user pick just the dash number is in my opinion not very good, because he would not know the dimensions unless he looks into the specs.

I think the selector should display the dash number, length and diameter, like `01 - 1 1/4 x 0.080in`, this way it should also appear ordered from shorter to longer length in the menu. If we use this approach then I believe it would be better to have values in inches in the CSV though.

hasecilu, 2024-02-03:
> * Then on the data panel, they could select the size (diameter parameter for threaded screws, but will be length for nails). _One problem here, some nail types like `common` have different subtypes (for common nails: Aluminum, Copper, Steel) so we could either:_

Are there relevant differences in the geometries? ISO norms also have different materials for some fasteners but geometries are the same (most of the times), ideally in the BOM designations the material should be included but as of now is not implemented.

> Tip style - Some nail types support different tip styles

I saw some tables named 01, 02, 03 etc. aren't these tables specific for those geometries features?

> Does this make sense?

I think it would be better to make dash number first. WHen people buy nails they do it using dash number or aren't?

tobiasfalk, 2024-02-03:
You mean DN Dia.xLen.(#42 1/5x3/8)?
That is defenetly a good way.

hasecilu, 2024-02-03:
> Yes, you're right, it's not possible to identify it by just the length then, but having the user pick just the dash number is in my opinion not very good, because he would not know the dimensions unless he looks into the specs.

The designer should know about the the norm is using

> You mean DN Dia.xLen.(#42 1/5inx3/8in)?

Could be possible to also include D and L on along the DN to people figure it out

tobiasfalk, 2024-02-03:
The Dia.xLen. style is standard for the DIN nails, so the combo box fill function would need to know when there is a DN

Anthony-Gaudino, 2024-02-03:
> Are there relevant differences in the geometries? ISO norms also have different materials for some fasteners but geometries are the same (most of the times), ideally in the BOM designations the material should be included but as of now is not implemented.

For different materials, there can be difference in the dimensions, for example the 1" copper common nail (F 1667 NLCMC) and the 1" steel common nail (F 1667 NLCMS) have different diameter shanks.

> I saw some tables named 01, 02, 03 etc. aren't these tables specific for those geometries features?
F 1667 NLBXA

Some nail types can have different features, for example, for `F 1667 NLBXA` it says:
`When specified, box nails shall have an altered or T-head with a diamond, blunt, or chisel point for use with power tools.`

And on  TABLE 21 Type I, Style 15—Flooring Nail it says: `Harded steel or steel wire, casing head or ﬂat-cupped countersunk head, diamond or blunt point, round, smooth or mechanically deformed shank`

So, there could be quite some variety of styles. I think the best would be to have separate geometries for head and shank / tip.
I don't know if it's easy to apply changes to geometry depending on a setting, for example, T-head, would be applying a cut to both sides of the nail.

I don't know how we would constrain some types, because it can be specific and don't allow changes.
`F 1667 NLCMS` for example just says:  `flat head, diamond point, round smooth shank` so geometry could not be changed, unless for details, like head pattern.

> I think it would be better to make dash number first. WHen people buy nails they do it using dash number or aren't?

When buying I don't think they do using dash number, usually the packaging says the penny size (2d, 6d...) and the length.
When the type of nail don't have a penny size, then in the packaging it shows the length.
As we saw in the standard, not all nails have a penny size specified, and some types have repeated length entries, so I think the best would be to show the dash number along with length and width so the user can easily pick the desired size.

hasecilu, 2024-02-05:
> Some nail types can have different features, ...

Ohh I see, a good example where "variants" on a single fastener exist is in PEMPressNut where `tcode` property is used to create geometry variations, so you need to create extra tables on csv files. You could add a property with a combobox if needed to choos tip type.

> so I think the best would be to show the dash number along with length and width so the user can easily pick the desired size.

Agreed

Anthony-Gaudino, 2024-02-07:
> Ohh I see, a good example where "variants" on a single fastener exist is in PEMPressNut where `tcode` property is used to create geometry variations, so you need to create extra tables on csv files. You could add a property with a combobox if needed to choose tip type.

Could you give an example of how should different types be created in the CSV?

If for example, the nail can have **diamond, blunt, or chisel** point and can have all types of head pattern, **no pattern, checkered, concentric ring**.

What if there are two tables in the CSV for two styles of nails, and one of them can have **diamond or blunt** points, but not **chisel** point?

hasecilu, 2024-02-09:
> Could you give an example of how should different types be created in the CSV?

You could have a `def` table (mandatory) where the main dimensions are there and create other `diamond`, `blunt`, `chisel` (optional) tables in the same file and only read those on their correspondant if statement.

I wouldn't worry much about heads because one thing we do not want is to make the models complex because that's unconvenient when you have a lot of fasteners.

If you want, create a branch in your fork and try creating a simple nail as the one I submitted, then try making some changes or adding data tables.

luzpaz, 2025-02-10:
Anything left to do here ?